micropython/micropython #16527: RFC: Use board name for firmware files

chrismas9, 2025-01-05:
### Description

Whenever I build firmware for a board I need to rename `firmware.elf`, etc to the board name before distributing it to users. Would it be useful to replace "firmware" with the board name in Makefile to automate this? I have tested it and the result looks like this:

```
LINK build-PYBV10/PYBV10.elf
   text    data     bss     dec     hex filename
 364920      52   26908  391880   5fac8 build-PYBV10/PYBV10.elf
GEN build-PYBV10/PYBV10-0.bin
GEN build-PYBV10/PYBV10-1.bin
GEN build-PYBV10/PYBV10.hex
GEN build-PYBV10/PYBV10.dfu
```

If interested I can submit a PR for the stm32 port, and probably other arm ports, but I don't have the build environment to do non-arm ports.

### Implementation

I intend to implement this feature and would submit a Pull Request if desirable

### Code of Conduct

Yes, I agree

Josverl, 2025-01-05:
I think this may be useful, and would argue that the VARIANT should be part of the file name if this is implemented.

Im pretty sure there are implications for the test and daily builds that need to be addressed as well.

Another alternative would be to use mpbuild, if mpbuild would offer a feature to rename the resulting firmware files  
 @mattytrentini ?

mattytrentini, 2025-01-05:
> Another alternative would be to use mpbuild, if mpbuild would offer a feature to rename the resulting firmware files @mattytrentini ?

Certainly could be done in mpbuild. May be better than trying to implement it in the various ports [c]make files.

However, personally, I _prefer_ having the build artifact always named the same, with the build directory providing the separation.

chrismas9, 2025-01-05:
My implementation does include the variant in FIRMWARE. I have decided to set FIRMWARE in `mpconfigboard.mk` where I have more control over the info in the filename. If there is no consensus I'm happy to keep this in my production branch.