micropython/micropython #16554: Multithread breaks file operations on pico2w.

tudorapo, 2025-01-08:
### Port, board and/or hardware

RPI_PICO2_W

### MicroPython version

MicroPython v1.25.0-preview.180.g495ce91ca on 2025-01-06; Raspberry Pi Pico 2 W with RP2350

### Reproduction

1. copy the below threads using code and run it:

```
import time
import _thread

global killme

def threadtest():
    global killme
    print("Other starts")
    while True:
        if killme:
            break
        print("másik thread")
        time.sleep(3.17)
    print("Other ends")

killme = False

try:
    new_thread = _thread.start_new_thread(threadtest, ())
    print("Main starts")
    while True:
        print("Main thread")
        time.sleep(1)
except KeyboardInterrupt as theInterrupt:
    print("Main ends")
    killme = True
```
2. run it
3. stop it
4. try to copy or delete any file with mpremote, it will hang
5. powercycle (pull out the usb conenctor and back) and things will work until I run the above code again.

### Expected behaviour

Expected to be able to upload code.

### Observed behaviour

Was not able to upload code. 

### Additional Information

When I copied/ran non-threaded code, a simple hello world the problem did not happen.

### Code of Conduct

Yes, I agree

jonnor, 2025-01-09:
Potentially the same issue as https://github.com/micropython/micropython/issues/13476

mendenm, 2025-01-10:
I had a lot of trouble with threads, much like what you are reporting, with inability to save files, or to restart the second thread, but finally have it to where I can ^C out of a program a lot like yours, and still have full control, including file access and restarts.  My skeleton program looked a lot like yours.

I don't know exactly what fixed it, but I think it may have had to do with a short sleep (1 second, probably much less would do) after you execute the `killme = True`.  I think exiting to REPL before the second thread has exited is trouble.  

To make my point as to the reliability of this suggestion, I have no idea.  I fiddled with a number of things, and now my code works right, with 100% repeatability.  What the change was, I am not exactly sure, but the sleep is nearly the only functional difference with your code.

YMMV