micropython/micropython #16563: ports/rp2: Pico 2 W with `PICO_PIO_USE_GPIO_BASE 1` and timer-based blink randomly causes all timers to stop

sfe-SparkFro, 2025-01-10:
### Port, board and/or hardware

Pico 2 W (actually tested with Pico Plus 2 W)

### MicroPython version

v1.24.0-181-g495ce91ca-dirty

### Reproduction

1. In `lib/pico-sdk/src/boards/include/boards/pico2_w.h`, add `#define PICO_PIO_USE_GPIO_BASE 1`
2. Build MicroPython for the Pico 2 W, then flash to the board
    * Tested with Pico Plus 2 W, but Pico 2 W probably also works
3. Run the code below
    * I'm using [mpremote](https://docs.micropython.org/en/latest/reference/mpremote.html) to copy/paste everything in paste mode (Ctrl+E)
4. The LED starts flashing, and `time.ticks_ms()` is printed every 100ms

```
from machine import Pin, Timer
import time

led = Pin("LED")

def cb1(foo):
    print(time.ticks_ms())
def cb2(foo):
    led.toggle()

timer1 = Timer(-1)
timer2 = Timer(-1)
timer1.init(freq=10, mode=Timer.PERIODIC, callback=cb1)
timer2.init(freq=10, mode=Timer.PERIODIC, callback=cb2)
```

### Expected behaviour

Both timers should continue to function indefinitely. The LED should continue blinking, and `time.ticks_ms()` should continue printing.

### Observed behaviour

After a random amount of time (a couple seconds to a couple minutes), the LED stops flashing and `time.ticks_ms()` stops printing. Both timers stop without any apparent reason. The REPL is still usable like normal.

### Additional Information

The reproduction steps above are a bit contrived, but it's actually causing problems for my real application; this was just a simple way to replicate the problem. I have a different board with an RP2350B, RM2, and LED connected to the RM2 (very similar to the Pico Plus 2 W). In my application, I have a handful of timers, one of which blinks the LED. After a short amount of time, all timers stop.

I have found that doing either of the following ensures all timers continue working:

- Disable the blink timer
    - This is not appropriate for my application.
- Add `#define PICO_PIO_USE_GPIO_BASE 0` to my board definition
    - This prevents me from using PIO on GPIO > 31.

I tried to replicate the problem in Pico SDK 2.1.0 without success. I ran the `picow_blink` example with the same `#define PICO_PIO_USE_GPIO_BASE 1` modification, but never observe any problems after ~30 minutes. So this appears to be a MicroPython bug.

### Code of Conduct

Yes, I agree

Gadgetoid, 2025-02-21:
AFAICT `PICO_PIO_USE_GPIO_BASE` doesn't do anything much in MicroPython apart from `ifdef` out some functions that otherwise aren't call. Aside from some bizarre alignment bug, I don't see a mechanism for this?

Are you sure it's this setting that's fixing things, versus you blowing away your build and producing a clean one?

sfe-SparkFro, 2025-03-10:
Thanks @Gadgetoid for looking into this! This appears to be resolved now.

I went back to commit `495ce91c` and was able to replicate the problem. I then updated to the latest stuff (keeping the `PICO_PIO_USE_GPIO_BASE` modification), and it seems to work fine now. Not sure exactly what the problem was, and I'm not going to hunt it down, I'm just glad it appears to be fixed now.

closed 2025-03-10 by sfe-SparkFro