micropython/micropython #16569: asyncio terminates unexpectedly

peterhinch, 2025-01-11:
### Port, board and/or hardware

Unix build (also verified on RP2 2)

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Raspberry Pi Pico2 with RP2350

### Reproduction

Paste the following at the REPL
```py
import asyncio

evt = asyncio.Event()
async def foo():
    data = [1,2,3]
    for d in data:
        print(d)
        await asyncio.sleep(1)
    await evt.wait()  # Should hang

async def main():
    await foo()

asyncio.run(main())
```

### Expected behaviour

Should print 1, 2, 3 then hang. This happens under CPython 3.10.12
```
1
2
3
```

### Observed behaviour

Under MP, after expected output, it returns to the REPL.
```
1
2
3
>>> 
```

### Additional Information

This may be related to https://github.com/micropython/micropython/issues/16318

I am encountering the phenomenon of premature return in a number of situations, notably in test scripts where there are no pending tasks. I have never observed it in applications where there are multiple concurrent tasks. 

### Code of Conduct

Yes, I agree

peterhinch, 2025-02-01:
To add a few observations.

The problem goes away if a second task is launched that simply does a repeated async sleep. This explains why the fault only occurs in simple scripts.

When the fault occurs the `run_until_complete` loop quits [here](https://github.com/micropython/micropython/blob/bfb1bee6feba9f06452a0e4b572ec4d15df325cf/extmod/asyncio/core.py#L168). It seems that while paused on
```py
    await evt.wait()  # Should hang
```
there is no `cur_task`, causing `run_until_complete` to quit prematurely.

While the example script is artificial I've encountered this bug a few times in simple scripts intended to test applications. The work-round is trivial, but the tendency for premature exit may confuse beginners.

Baker68, 2025-02-10:
Also present on MicroPython v1.24.0 on ESP32

dpgeorge, 2025-05-07:
Fixed by bdb7e036d2400c25beeb2b4252ff51c5255c4020

closed 2025-05-07 by dpgeorge