micropython/micropython #16579: rp2: asm_pio_encode() sideset/delay faulty logic

melberi, 2025-01-13:
### Port, board and/or hardware

RP2040

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Raspberry Pi Pico with RP2040

### Reproduction

In a PIO instruction, 5 bits are reserved for sideset/delay. Using one or more sideset pins reduces the amount of bits left for delay. However, the asm_pio_encode() complains about delay too large with the following invocation:

rp2.asm_pio_encode('mov(x,y).side(1) [7]', 2, sideset_opt=True)

The second argument to the function is sideset count and is inclusive of the sideset enable bit which is required when sideset is optional. In this example two bits are reserved for sideset purposes and three are left for delay allowing a maximum of 7.

### Expected behaviour

The function is expected to return 48930 equivalent to 0b1011111100100010. 

Note the sideset/delay in bits 8-12 (counting up from 0), 11111, with the two most significant bits being the sideset enable and the actual sideset bit and the rest of the bits signifying a delay value of 7. 

### Observed behaviour

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "rp2.py", line 298, in asm_pio_encode
  File "<string>", line 1, in <module>
  File "rp2.py", line 79, in __getitem__
  File "rp2.py", line 84, in delay
PIOASMError: delay too large

### Additional Information

I came across this bug when figuring out why the asm_pio_encode() function would not seem to work correctly with sideset count equal to 1 while sideset_opt=True. 

The documentation did not mention about sideset count being inclusive of sideset enable bit. That information was found in the codebase as a comment. It follows from the SIDESET_COUNT count in SM#_PINCTRL register of RP2040 where the sideset enable bit is included in the count. 

### Code of Conduct

Yes, I agree

melberi, 2025-01-14:
Yes and the bug does propagate to that function as well.

To be clear, the issue is that when two bits are being used for sideset purposes, the asm_pio_encode() behaves like three bits were used leaving only two bits for delay. In reality, three bits should be available for delay.

Gadgetoid, 2025-01-14:
Note the source for this function is available as Python code here: https://github.com/micropython/micropython/blob/1b4c969ce0c2ed83bcf366e1b6d977df3d72a9d2/ports/rp2/modules/rp2.py#L276-L302

It does appear as though there's a difference between creating a `PIOASMEmit` and poking values into it for a single instruction encode versus setting it up properly for a full multi-pass encode.

For a single instruction, `start_pass` is never called, and that does some finagling with `sideset_count` if `sideset_opt` is `True` - https://github.com/micropython/micropython/blob/1b4c969ce0c2ed83bcf366e1b6d977df3d72a9d2/ports/rp2/modules/rp2.py#L68-L72