micropython/micropython #16589: RMT receive

Zibri, 2025-01-15:
### Description

I just found out that microphthon can on send RMT but not receive. That's a shame which forces me to use nodemcu while I prefer microphthon by a long shot. RMT receive is very important and without it, receiving pulses is way more inaccurate.
In my project I need to determine the pulse length accurately. Please implement RMT receive!

### Code Size

Should not be so big. But honestly I don't know.

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

0wwafa, 2025-01-16:
The reason:

def test():
     s=time.ticks_cpu()
     time.sleep(1)
     print(time.ticks_diff(time.ticks_cpu(),s))
     
 the reults will have the overhead of the code which is around 4700 nanoseconds (clock 20-120) and more than 8000 at 240 mhz.
 
 The same happens if you sent an IRQ on a pin falling (or raising) edge.

RMT instead will return the time of each edge (or more couples) each time is called and has a resolution up to 12.5 nanoseconds!
It would be amazing to be able to use it without ditching micropython and coding in C.

closed 2025-02-20 by mattytrentini

elvis-epx, 2025-12-11:
Proposal in https://github.com/micropython/micropython/pull/18454