micropython/micropython #16601: Please add support for EPS32-CAM

0wwafa, 2025-01-17:
### Description

[image]

chip detail

[image]

I know there is a "driver" on github but it's from 2022. 

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

0wwafa, 2025-01-17:
As far as I know it is supported in circuitpython, but I like micropython way more.

NickInManchester, 2025-01-23:
I'd support this. There are several Esp32s3 boards with camera attachments. Using the latest stable micropython with Esp-Idf 5.0 I've tried compiling and solved several problems in the. mk files and make includes before getting stuck.

For all the ports etc the make uses Ninja to generate command files. I so far failed to find how to add one or two missing include files. The make and ordinary make files etc are almost undocumented and are vast. Verbose mode doesn't help much. Long commands end ... So not everything is visible. I looked for a flag to avoid this but I've not found one yet.

If someone who understands the make system could configure the Cam driver it would be very helpful for my projects.

Thanks in advance 

Nick

NickInManchester, 2025-01-23:
Or simply give some advice on make and/or how Ninja is used in the make systems for headers and ports....?

Thanks

Nick

jonnor, 2025-03-03:
The ESP32 generic firmware will work great on these boards. It will not contain a camera driver, but basically everything else is there. For the camera, there an excellent external C module available at https://github.com/cnadler86/micropython-camera-API

There has been some discussions on standardizing such an API and driver. Ref https://github.com/orgs/micropython/discussions/15985

Let us continue the discussions there? Proposing this issue as duplicate

andrewleech, 2025-03-04:
The problem with the ESP32-CAM board is there's lots of slightly different ones that essentially look / branded similarly, so it can be difficult to "officially" support with a board profile.

I've got a couple of them that look identical, however it turns out the base board (with the usb serial converter) is different for each, for one of them any time the base board is plugged in it's forced into update mode all the time regardless of pushbutton presses, so that base board can't be used for repl. The other base board I've got never jumps to update mode, the pushbuttons don't work for that either, nor does the esp auto-jump to bootloader.

That being said yeah you can use a generic firmware with spiram enabled and it should work fine if you can program it.  Building your own with the camera driver linked by @jonnor should be fairly straightforward with `mpbuild` however I don't think there are any instructions yet for how to pass the make args to bring in the user module. That project does provide pre-built firmware you can try though!