micropython/micropython #16605: Pin irq firing wildly on ESP32

0wwafa, 2025-01-17:
### Port, board and/or hardware

ESP32 Wroom32

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Generic ESP32 module with ESP32

### Reproduction

```
def irq_handler(pin):
     print("got irq")

input_pin = Pin(0) # the button on ESP32
input_pin.irq(trigger=Pin.IRQ_FALLING, handler=irq_handler)

```

### Expected behaviour

Pressing the button I should see "got irq" once as I press it.

### Observed behaviour

I get one irq as I press it and 8-10 irqs when I release it.

I tried using RISING and I get the same result.

### Additional Information

Perhaps there is something I did not understand... but when I press the button I have ONE falling and one rising edges. (or the opposite?)

Anyway they are 2.

Why do I get so many?

### Code of Conduct

Yes, I agree

0wwafa, 2025-01-17:
An even weirder result if you do:

```
from machine import Pin
def irq_handler(pin):
     print("got irq", pin.value())

input_pin = Pin(0,Pin.IN) # the button on ESP32
input_pin.irq(trigger=Pin.IRQ_FALLING, handler=irq_handler)
```

if you press the button you will see many IRQs and sometimes the value is 1!

dhalbert, 2025-01-17:
Mechanical buttons have contact bounce. You'll need to do some kind of software or electrical debouncing.

0wwafa, 2025-01-17:
> Mechanical buttons have contact bounce. You'll need to do some kind of software or electrical debouncing.

I don't really need it for this purpose... I need to measure the length of a full wave ON/OFF   of a pulse.

So I just take the time from 2 falling edges.
But I am getting multiple firings even in that case.

andrewleech, 2025-01-17:
What's generating the pulse? Even other circuits that make pulse outputs can have ringing in the edges that need to be filtered out they'll trigger an input irq multiple times.

If you need to accurately measure pulse width you might need a bit of analog filtering on the signal (like a basic RC on the input pin).

0wwafa, 2025-01-17:
Ok I made another test: I connected pin 13 and VIN (last pin on the same row)

then I manually make the connection between the 2 three times.

The irq fired 69 times!

I don't really know anymore how to use micropython to measure a signal reliably.

I can accept I make a false contact perhaps (but I don't) and I wouldn't have been surpised if it fired like 4-5 times instead of 3.

BUT 69?

smhodge42, 2025-01-18:
Have you heard of switch bounce?

andrewleech, 2025-01-18:
> then I manually make the connection between the 2 three times.
> 

Do you mean connecting a wire / jumper between the pins by hand? This is the same as a mechanical pushbutton and definitely has switch bounce. I'd often expect hundreds of irq firings in a situation like this!

Here's an article about it if your want to know more https://www.allaboutcircuits.com/technical-articles/switch-bounce-how-to-deal-with-it/

robert-hh, 2025-01-18:
It seems that all was said already, but not by everyone. So I add my 5 cents.
There is another phenomenon besides the contact bounce of a switch. That is the undefined state of an input in the transition between `solid low` and `solid high`, usually in the region of Vcc/2. You did not tell how you have set up your test, and if usin single button as switch, what the value of the pull-up (or pull-down) resistor was. If that one has a high value, the input is in the Vcc/2 region for a quite long time, at which the value taken by the input logic can alter between 0 and 1 quite many times. Below is a oscilloscope screenshot illuminating this. The green trace is the input signal, a triangle signal. The yellow trace shows the value read by the MCU. The top trace pair shows the full time, the lower trace pair the Zoom. You see, that value read changes a lot in the transition area. That is not a bug, but typical for digital circuits. It is up to the one setting up the circuit to keep transition times low and suppress contact bouncing.

[image]

0wwafa, 2025-01-18:
Ok. I was testing my program before connecting it to a device... so when I will connect it to a device I will have only ONE rising irq and one falling?

robert-hh, 2025-01-18:
That is to be expdxtdd, when the fise and fall times of the signal are sufficiently low.

robert-hh, 2025-01-18:
You can check that by using another Pin of the ES32 as stimulus, like:
```
from machine import Pin
from time import sleep
def irq_handler(pin):
     print("got irq", pin.value())

output_pin = Pin(5, Pin.OUT, value=0)
input_pin = Pin(4,Pin.IN) # 
input_pin.irq(trigger=Pin.IRQ_FALLING|Pin.IRQ_RISING, handler=irq_handler)

output_pin(1);sleep(1);output_pin(0)
```
Note: avoid using Pin 0 as input. I Pin 0 at an ESP32 is low on reset, the device will switch to bootloader mode.

0wwafa, 2025-01-18:
> avoid using Pin 0 as input.

it was just for testing the routine... I will sure use different pins (13 for input and 14 for output) after the main program has been tested.

0wwafa, 2025-01-18:
> There is another phenomenon besides the contact bounce of a switch [...]

Thank you very much for the explanation.

closed 2025-01-19 by robert-hh