micropython/micropython #16618: Not able to build firmware for rpi pico 2w

ukrolelo, 2025-01-21:
### Port, board and/or hardware

rp2350

### MicroPython version

Latest Micropython as of today 21.1.2025

### Reproduction

make -j 16 BOARD=RPI_PICO2_W

### Expected behaviour

finish build

### Observed behaviour

`
root@myca:~/micropython/micropython/ports/rp2# make BOARD=RPI_PICO2_W clean
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -f -rf build-RPI_PICO2_W
root@myca:~/micropython/micropython/ports/rp2# make BOARD=RPI_PICO2_W submodules
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make -f ../../py/mkrules.mk GIT_SUBMODULES="lib/pico-sdk" submodules
make[1]: Entering directory '/root/micropython/micropython/ports/rp2'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/pico-sdk
Synchronizing submodule url for 'lib/pico-sdk'
usage: git submodule [--quiet] [--cached]
   or: git submodule [--quiet] add [-b <branch>] [-f|--force] [--name <name>] [--reference <repository>] [--] <repository> [<path>]
   or: git submodule [--quiet] status [--cached] [--recursive] [--] [<path>...]
   or: git submodule [--quiet] init [--] [<path>...]
   or: git submodule [--quiet] deinit [-f|--force] (--all| [--] <path>...)
   or: git submodule [--quiet] update [--init] [--remote] [-N|--no-fetch] [-f|--force] [--checkout|--merge|--rebase] [--[no-]recommend-shallow] [--reference <repository>] [--recursive] [--[no-]single-branch] [--] [<path>...]
   or: git submodule [--quiet] set-branch (--default|--branch <branch>) [--] <path>
   or: git submodule [--quiet] set-url [--] <path> <newurl>
   or: git submodule [--quiet] summary [--cached|--files] [--summary-limit <n>] [commit] [--] [<path>...]
   or: git submodule [--quiet] foreach [--recursive] <command>
   or: git submodule [--quiet] sync [--recursive] [--] [<path>...]
   or: git submodule [--quiet] absorbgitdirs [--] [<path>...]
make[1]: Leaving directory '/root/micropython/micropython/ports/rp2'
make[1]: Entering directory '/root/micropython/micropython/ports/rp2'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/mbedtls lib/tinyusb lib/lwip lib/btstack lib/cyw43-driver lib/micropython-lib
Synchronizing submodule url for 'lib/btstack'
Synchronizing submodule url for 'lib/cyw43-driver'
Synchronizing submodule url for 'lib/lwip'
Synchronizing submodule url for 'lib/mbedtls'
Synchronizing submodule url for 'lib/micropython-lib'
Synchronizing submodule url for 'lib/tinyusb'
usage: git submodule [--quiet] [--cached]
   or: git submodule [--quiet] add [-b <branch>] [-f|--force] [--name <name>] [--reference <repository>] [--] <repository> [<path>]
   or: git submodule [--quiet] status [--cached] [--recursive] [--] [<path>...]
   or: git submodule [--quiet] init [--] [<path>...]
   or: git submodule [--quiet] deinit [-f|--force] (--all| [--] <path>...)
   or: git submodule [--quiet] update [--init] [--remote] [-N|--no-fetch] [-f|--force] [--checkout|--merge|--rebase] [--[no-]recommend-shallow] [--reference <repository>] [--recursive] [--[no-]single-branch] [--] [<path>...]
   or: git submodule [--quiet] set-branch (--default|--branch <branch>) [--] <path>
   or: git submodule [--quiet] set-url [--] <path> <newurl>
   or: git submodule [--quiet] summary [--cached|--files] [--summary-limit <n>] [commit] [--] [<path>...]
   or: git submodule [--quiet] foreach [--recursive] <command>
   or: git submodule [--quiet] sync [--recursive] [--] [<path>...]
   or: git submodule [--quiet] absorbgitdirs [--] [<path>...]
make[1]: Leaving directory '/root/micropython/micropython/ports/rp2'
root@myca:~/micropython/micropython/ports/rp2# make BOARD=RPI_PICO2_W -j 16
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
[ -e build-RPI_PICO2_W/Makefile ] || cmake -S . -B build-RPI_PICO2_W -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=RPI_PICO2_W -DMICROPY_BOARD_DIR="/root/micropython/micropython/ports/rp2/boards/RPI_PICO2_W"
PICO_SDK_PATH is /root/micropython/micropython/lib/pico-sdk
Target board (PICO_BOARD) is 'pico2_w'.
Using board configuration from /root/micropython/micropython/lib/pico-sdk/src/boards/include/boards/pico2_w.h
[... 654 lines cut ...]
 ^
/usr/lib/gcc/arm-none-eabi/8.3.1/include/arm_acle.h:223:1: error: target CPU does not support ARMv8-M Security Extensions
/usr/lib/gcc/arm-none-eabi/8.3.1/include/arm_acle.h:230:1: error: target CPU does not support ARMv8-M Security Extensions
 {
 ^
/usr/lib/gcc/arm-none-eabi/8.3.1/include/arm_acle.h:230:1: error: target CPU does not support ARMv8-M Security Extensions
make[3]: *** [CMakeFiles/firmware.dir/build.make:3304: CMakeFiles/firmware.dir/pendsv.c.o] Error 1
make[3]: *** [CMakeFiles/firmware.dir/build.make:3265: CMakeFiles/firmware.dir/mpnetworkport.c.o] Error 1
make[3]: Leaving directory '/root/micropython/micropython/ports/rp2/build-RPI_PICO2_W'
make[2]: *** [CMakeFiles/Makefile2:2221: CMakeFiles/firmware.dir/all] Error 2
make[2]: Leaving directory '/root/micropython/micropython/ports/rp2/build-RPI_PICO2_W'
make[1]: *** [Makefile:103: all] Error 2
make[1]: Leaving directory '/root/micropython/micropython/ports/rp2/build-RPI_PICO2_W'
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:63: all] Error 1

`

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-01-21:
> The C compiler identification is GNU 8.3.1

It looks like your C compiler is too old.  You'll need to update it.

closed 2025-01-21 by dpgeorge

ukrolelo, 2025-01-22:
@dpgeorge I would like to say thank you! you saved me many hours ! <3 <3 <#