micropython/micropython #16624: Please update the version of cyw43-driver

edmcman, 2025-01-23:
### Description

Please update the version of cyw43-driver to include [this commit](https://github.com/georgerobotics/cyw43-driver/commit/c1075d4bc440422cf2b2fd12c64a1f53f77660ee) which fixes an issue I encountered in micropython (https://github.com/micropython/micropython/issues/16482).

### Code Size

N/A.  The code size difference should be minimal.

### Implementation

I intend to implement this feature and would submit a Pull Request if desirable, I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

dpgeorge, 2025-01-23:
Yes, will do this for the upcoming 1.25.0 release.

dpgeorge, 2025-03-12:
See #16914 for the update of the driver.

closed 2025-03-13 by dpgeorge