micropython/micropython #16627: rp2: Chained DMA breaking CYW43 wireless

Gadgetoid, 2025-01-23:
### Port, board and/or hardware

Pico2 W

### MicroPython version

MicroPython v1.25.0-preview.221.gc69f0e4ee on 2025-01-23; Raspberry Pi Pico 2 W with RP2350

### Reproduction

Run the following code on a Pico2 W.

`wifi.connect(...)` will fail with an EPERM.

Comment out `dma_ctrl.active(True)` and it will succeed.

Note: Power cycle between attempts, since once the WiFi is fired up it tends to be happier.

```python
import rp2
import time
import uctypes
from array import array
import network

dma = rp2.DMA()
dma_ctrl = rp2.DMA()

# A sufficiently large source bitstream
bitstream = bytearray(64 * 1024)
bitstream_addr = uctypes.addressof(bitstream)

# These are the bytes that the control DMA will write to its chain target
control_bytes = array("L")
control_bytes.append(bitstream_addr)  # CH0_READ_ADDR

# This is the target address of the control DMA (our first DMA channels read address)
CH0_READ_ADDR = uctypes.addressof(dma.registers[0:])

# Dummy bytes to give our DMA transfer a nondescript target
target_bytes = array("L")
target = uctypes.addressof(target_bytes)

dma.config(
    read = bitstream_addr,
    write = target_bytes,
    count = len(bitstream) // 4,
    ctrl = dma.pack_ctrl(
        size      = 2,  # 0 = byte, 1 = half word, 2 = word
        irq_quiet = True,
        inc_read = True,
        inc_write = False,
        bswap     = False,
        chain_to  = dma_ctrl.channel,
    ),
    trigger = False
)

dma_ctrl.config(
    read  = control_bytes,
    write = CH0_READ_ADDR,
    count = 1,
    ctrl  = dma_ctrl.pack_ctrl(
        size      = 2,  # 0 = byte, 1 = half word, 2 = word
        inc_read = False,
        inc_write = False,
        irq_quiet = True,
        chain_to  = dma.channel, # since the register write is non triggering
    ),
    trigger = False
)

# Start the control channel
# Comment this out to get a "PASS"
dma_ctrl.active(True)

# Set up wireless
wlan = network.WLAN(network.STA_IF)
wlan.active(True)

# Try to connect. SSID/PSK don't matter.
# This will fail with EPERM
try:
    wlan.connect("test", "test")
    print("PASS!")
except OSError as e:
    print(e)
    print("FAIL!")
finally:
    # Teardown
    dma.active(False)
    dma.close()
    dma_ctrl.active(False)
    dma_ctrl.close()

```

### Observed behaviour

This is a pure MicroPython reproduction of an issue I discovered with our C++ USER_C_MODULE drivers for our Unicorn boards.

It seems to be impossible to fire up WiFi after firing up the display. I don't have a good sense for why.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

MichaelBell, 2025-01-23:
This looks to be an interaction with the DMA channel aborts in the CYW43 driver.  This patch to pico-sdk fixes it for me:

```
diff --git a/src/rp2_common/pico_cyw43_driver/cyw43_bus_pio_spi.c b/src/rp2_common/pico_cyw43_driver/cyw43_bus_pio_spi.c
index bcc7284..1c6bd84 100644
--- a/src/rp2_common/pico_cyw43_driver/cyw43_bus_pio_spi.c
+++ b/src/rp2_common/pico_cyw43_driver/cyw43_bus_pio_spi.c
@@ -260,6 +260,7 @@ int cyw43_spi_transfer(cyw43_int_t *self, const uint8_t *tx, size_t tx_length, u
         pio_sm_exec(bus_data->pio, bus_data->pio_sm, pio_encode_jmp(bus_data->pio_offset));
         dma_channel_abort(bus_data->dma_out);
         dma_channel_abort(bus_data->dma_in);
+        asm volatile (" nop\n nop\n nop " );
 
         dma_channel_config out_config = dma_channel_get_default_config(bus_data->dma_out);
         channel_config_set_bswap(&out_config, true);
@@ -301,6 +302,7 @@ int cyw43_spi_transfer(cyw43_int_t *self, const uint8_t *tx, size_t tx_length, u
         pio_sm_exec(bus_data->pio, bus_data->pio_sm, pio_encode_out(pio_y, 32));
         pio_sm_exec(bus_data->pio, bus_data->pio_sm, pio_encode_jmp(bus_data->pio_offset));
         dma_channel_abort(bus_data->dma_out);
+        asm volatile (" nop\n nop\n nop " );
 
         dma_channel_config out_config = dma_channel_get_default_config(bus_data->dma_out);
         channel_config_set_bswap(&out_config, true);
```

MichaelBell, 2025-01-24:
Yep, I did try that and it didn't help.

What I haven't tried is polling the abort register for the abort completion, which is a new thing on RP2350, and feels like it might be the correct thing to do?  Although potentially even if that worked it might just be because the check took enough cycles, which is presumably all the `nops` are doing.  Note the three `nops` was just my first guess, not anything scientifically determined.

But I think this needs looking at by someone that understands what is going on under the hood, rather than speculative fixes!

starblue, 2025-01-24:
According to the data sheet `dma_channel_abort` should wait for the bit in register `CHAN_ABORT` to become 0 again after writing a 1 (there's even a comment to that effect in the code). But it only waits on a completely different busy bit. https://github.com/raspberrypi/pico-sdk/blob/95ea6acad131124694cda1c162c52cd30e0aece0/src/rp2_common/hardware_dma/include/hardware/dma.h#L544

MichaelBell, 2025-01-24:
Yes, that change was a workaround for an RP2040 errata: https://github.com/raspberrypi/pico-sdk/pull/816/files
I wonder if RP2350 should have the original implementation...

Gadgetoid, 2025-01-25:
I put together a pure C test case, and both that and MicroPython seem to be fixed by this patch: https://github.com/raspberrypi/pico-sdk/pull/2209

I can't attest to understand what's going on and why this works, and why it is not an issue on the RP2040 but:

1. Yay. A fix!
2. Ohno, how can we get this into MicroPython short of downstream backport style patches?

Gadgetoid, 2025-01-29:
Note: this patch is now in devel and closed 3 issues relating to cyw43 instability.

CraSmeets, 2025-02-05:
Even with the patch (https://github.com/raspberrypi/pico-sdk/pull/2209) I still get the OSError _EPERM_ when I run wlan.active(True) with a second thread active. This only seems to happen on my Pico2 W board and not Pico1.

Test code:

```
import network
import time
import _thread

SSID = ""
password = ""

def connect_normal():
    print("Connecting to network")

    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)

    wlan.connect(SSID, password)

    start_time = time.time()
    while not wlan.isconnected() and time.time() - start_time < 100:
        print("Waiting for connection...")
        time.sleep(2)

    if wlan.isconnected():
        print("Connected to network")
    else:
        print("Failed to connect to network")
    print('network config:', wlan.ifconfig())

    print("Sleep for 2 seconds")
    time.sleep(2)
    wlan.disconnect()

def main_thread():
    connect_normal()

def second_thread():
    for (i) in range(5):
        print("Second thread-")
        time.sleep(2)

_thread.start_new_thread(second_thread, ())
main_thread()
```
Resulting in:

```
[CYW43] F2 not ready
Traceback (most recent call last):
  File "<stdin>", line 40, in <module>
  File "<stdin>", line 32, in main_thread
  File "<stdin>", line 14, in connect_normal
OSError: [Errno 1] EPERM
```

I do not get this error when you switch around the threads, `main_thread` in `start_new_thread()`, or when you do activate Wlan before you start the thread. However, in both situations, it never connects for me while commenting on the line `_thread.start_new_thread(second_thread, ())` results almost immediately in a connection.

Let me know if the never connecting in a thread is relevant for the EPERM error or if I should start a new issue for it.

dpgeorge, 2025-02-06:
> Note: this patch is now in devel and closed 3 issues relating to cyw43 instability.

Yes, that's great.  But I don't know how we can get it here without waiting for a pico-sdk release.

Gadgetoid, 2025-02-06:
> Yes, that's great. But I don't know how we can get it here without waiting for a pico-sdk release.

I'm currently backporting by saving the relevant patch files and using CI to apply them... I don't think I'd suggest that approach here. I'm afraid waiting is probably the only reasonable thing to do.

At least in the interim it's documented and producing a patched build is pretty trivial for anyone who cares to do so.