micropython/micropython #16640: WASM example not showing most properties from `fetch()` call

benfry, 2025-01-24:
### Port, board and/or hardware

WebAssembly

### MicroPython version

MicroPython tag v1.24.1 built from source.

### Reproduction

Using the example from <https://github.com/micropython/micropython/tree/master/ports/webassembly>

```html
<!doctype html>
<html>
  <head>
    <script src="build-standard/micropython.mjs" type="module"></script>
  </head>
  <body>
    <pre id="micropython-stdout"></pre>
    <script type="module">
      const stdoutWriter = (line) => {
        document.getElementById("micropython-stdout").innerText += line + "\n";
      };
      const mp = await loadMicroPython({stdout:stdoutWriter});
      await mp.runPythonAsync(`
        import js
        url = "https://api.github.com/users/micropython"
        print(f"fetching {url}...")
        res = await js.fetch(url)
        json = await res.json()
        for i in dir(json):
          print(f"{i}: {json[i]}")
      `);
    </script>
  </body>
</html>
```

I'm getting this as the result in the browser:

```
fetching https://api.github.com/users/micropython...
__class__: <undefined>
id: 6298560
type: Organization
__del__: <undefined>
name: MicroPython
new: <undefined>
url: https://api.github.com/users/micropython
```
…where it's only including `id`, `type`, `name`, and `url` from the JSON file that's fetched. 

### Expected behaviour

I'm expecting everything else from the JSON blob returned by <https://api.github.com/users/micropython>, i.e.:

```json
"login": "micropython",
"id": 6298560,
"node_id": "MDEyOk9yZ2FuaXphdGlvbjYyOTg1NjA=",
"avatar_url": "https://avatars.githubusercontent.com/u/6298560?v=4",
"gravatar_id": "",
"url": "https://api.github.com/users/micropython",
"html_url": "https://github.com/micropython",
"followers_url": "https://api.github.com/users/micropython/followers",
"following_url": "[https://api.github.com/users/micropython/following{/other_user}](https://api.github.com/users/micropython/following%7B/other_user%7D)",
"gists_url": "[https://api.github.com/users/micropython/gists{/gist_id}](https://api.github.com/users/micropython/gists%7B/gist_id%7D)",
"starred_url": "[https://api.github.com/users/micropython/starred{/owner}{/repo}](https://api.github.com/users/micropython/starred%7B/owner%7D%7B/repo%7D)",
"subscriptions_url": "https://api.github.com/users/micropython/subscriptions",
"organizations_url": "https://api.github.com/users/micropython/orgs",
"repos_url": "https://api.github.com/users/micropython/repos",
"events_url": "[https://api.github.com/users/micropython/events{/privacy}](https://api.github.com/users/micropython/events%7B/privacy%7D)",
"received_events_url": "https://api.github.com/users/micropython/received_events",
"type": "Organization",
"user_view_type": "public",
"site_admin": false,
"name": "MicroPython",
"company": null,
"blog": "https://micropython.org/",
"location": null,
"email": null,
"hireable": null,
"bio": "The MicroPython project",
"twitter_username": null,
"public_repos": 15,
"public_gists": 0,
"followers": 1236,
"following": 0,
"created_at": "2014-01-01T19:12:11Z",
"updated_at": "2020-04-08T04:50:46Z"
```

### Observed behaviour

Doing `print('received_events_url')` shows the correct output, but it appears that `dir()` is only listing a few keys? 

Is there another way to get a list of all the keys or is this just not possible? 

### Additional Information

Just getting started so I may be missing something entirely. I didn't have much luck finding other `JsProxy` documentation outside of just reading the source. 

Thanks!

### Code of Conduct

Yes, I agree

dpgeorge, 2025-01-25:
Thanks for the report.  I can reproduce the issue.

I'm not sure exactly what's wrong, but the JSON result from the fetch is a JavaScript Object and it's something to do with the way Objects are proxied to MicroPython that is giving the issue.

> Is there another way to get a list of all the keys or is this just not possible?

Yes, you can do this:
```py
        for i in js.Object.keys(json):
          print(f"{i}: {json[i]}")

```
Or if you want a Python list of the keys:
```py
keys = list(js.Object.keys(json))
```

benfry, 2025-01-25:
Got it; appreciate the fast response and workaround!