micropython/micropython #16644: ports/stm32: Add memory map with attributes for different MCU families.

iabdalkader, 2025-01-25:
### Description

There's a general DMA/Cache-management issue with the MicroPython's stm32 port; most drivers make assumptions about the memory (cachable or not DAM-acessible or not) which probably works fine for all (most?) of the default linker scripts, which happen to place everything in one memory region, but when buffers are moved around, drivers starts failing.

A few examples to demonstrate the issue (there are many more in sdcard, I2C etc..):

1. DAC should flush buffer before DMA call *if memory is cacheable* and fail if it's not DMA-accesible: https://github.com/micropython/micropython/blob/a4ab847688406bb31eb531e07155bf173a8785aa/ports/stm32/dac.c#L201

2. SPI regular transfer (fallback) path should be taken *if the memory is not accessible by DMA*: https://github.com/micropython/micropython/blob/a4ab847688406bb31eb531e07155bf173a8785aa/ports/stm32/spi.c#L588
For example:
```c
if (len == 1 || query_irq() == IRQ_STATE_DISABLED || !dma_is_accessible(src)) {
```
It should also Not do cache-management if memory is Not cacheable.

The problem will only get worse as we add more MCUs with more memory regions. To fix this issue, I propose we implement some sort of memory map per MCU family, with attributes per region: cacheable, DAM-accessible, TCM, external etc.. And a new generic API to query those attributes. It would be great if this mechanism is generic somehow, so it can be reused by future ports with more complex memory. I'd be very happy to work on this, but I need some guidance with the API.

### Implementation

I intend to implement this feature and would submit a Pull Request if desirable

### Code of Conduct

Yes, I agree

kwagyeman, 2025-01-25:
This is cool and would be very useful.

iabdalkader, 2025-01-27:
@dpgeorge Any thoughts?

Bortania, 2025-03-01:
@iabdalkader can you point out on the "patches to all of those driver" that you mentioned in your post.
Here we have an issue with DAC buffer on nucleo F7 board. Maybe the reason is memory cache.

iabdalkader, 2025-03-01:
> can you point out on the "patches to all of those driver" that you mentioned in your post.

Those patches are not applicable for all cases, otherwise I would have sent them as fixes. For example DAC uses DMA, in `timed` mode, so the memory needs to be cleaned/flushed before DMA starts. But the memory could also be TCM, in this case it should Not be flushed.
The point is, the memory view in MicroPython is overly simplistic and assumes that any provided memory will just work. Most of the chips/boards just happen to use memory that works in most cases, but once you start customizing the linker script and moving buffers around, drivers start failing.

Bortania, 2025-03-01:
> Those patches are not applicable for all cases, otherwise I would have sent them as fixes. For example DAC uses DMA, in `timed` mode, so the memory needs to be cleaned/flushed before DMA starts. But [...]

Exactly, here we have the issue with dac.write_timed(dacBuf, dacTimer, mode=dac.CIRCULAR) for the Nucleo STM32F767ZI board.
Is this issue solvable in micropython?

iabdalkader, 2025-03-01:
> Is this issue solvable in micropython?

Yes, it is. This patch may fix it for you, but it's not a good long-term solution. The proper fix is this feature proposed here.

```C
diff --git a/ports/stm32/dac.c b/ports/stm32/dac.c
index 54d5acc6c..dcbd93516 100644
--- a/ports/stm32/dac.c
+++ b/ports/stm32/dac.c
@@ -198,6 +198,10 @@ static void dac_start_dma(uint32_t dac_channel, const dma_descr_t *dma_descr, ui
     #endif
     }
 
+    #ifdef __DCACHE_PRESENT
+    MP_HAL_CLEAN_DCACHE(buf, len);
+    #endif
+
     dma_nohal_deinit(dma_descr);
     dma_nohal_init(dma_descr, DMA_MEMORY_TO_PERIPH | dma_mode | dma_align);
     dma_nohal_start(dma_descr, (uint32_t)buf, base + dac_align, len);
```

Bortania, 2025-03-03:
> > Is this issue solvable in micropython?
> 
> Yes, it is. This patch may fix it for you, but it's not a good long-term solution. The proper fix is this feature proposed here.
> 
> diff --git a/ports/stm32/dac.c b/ports/stm32/dac.c
> index 54d5acc6c..dcbd93516 100644
> --- a/ports/stm32/dac.c
> +++ b/ports/stm32/dac.c
> @@ -198,6 +198,10 @@ static void dac_start_dma(uint32_t dac_channel, const dma_descr_t *dma_descr, ui
>      #endif
>      }
>  
> +    #ifdef __DCACHE_PRESENT
> +    MP_HAL_CLEAN_DCACHE(buf, len);
> +    #endif
> +
>      dma_nohal_deinit(dma_descr);
>      dma_nohal_init(dma_descr, DMA_MEMORY_TO_PERIPH | dma_mode | dma_align);
>      dma_nohal_start(dma_descr, (uint32_t)buf, base + dac_align, len);

@iabdalkader thank you for reply.
Do you know if issue with cache can influence SPI? I see that there was commit to fix for cache and SPI in MicroPython source code. But   when I establis communicate of DUT with nucleo F7 board through SPI, SPI reading is 0x0.
Did you have some similar experience with SPI?

iabdalkader, 2025-03-03:
> Do you know if issue with cache can influence SPI?

Every driver here can be affected by cache or other memory related issues. That's what I'm trying to fix with this new feature.

Gadgetoid, 2025-03-28:
I think the crucial aspect of this, and perhaps the most glossed over portion, is the proposal for the ability to alloc memory from a pool with the given attributes.

Since an SRAM/PSRAM hybrid (we're working toward this on RP2) is, from the perspective of userspace, a total lottery (caveat that if you alloc early you'll probably be in SRAM if there's enough and if it's up-front) then there's nothing the end-user can really *do* if a driver *requires* DMA-accessible RAM, fast RAM, page-aligned RAM etc but expects a `bytearray` created and supplied by the user.

Can we get away with adding new syntax for this?

Allocation decorators? 😆 

```python
@memory.SRAM(align=1024)
buffer = bytearray(1024)

@memory.any(align=1024, dma=True)
buffer = bytearray(1024)
```

Or a context manager?

```python
with memory.any(align=1024, dma=True):
    buffer1 = bytearray(1024)
    buffer2 = bytearray(1024)
```

iabdalkader, 2025-03-28:
I haven't thought about exposing this functionality to Python code, I was mainly concerned with C code/drivers, but I can see how this might be very useful for Python drivers, or even regular code that requires a small amount of fast memory. As for syntax, I personally prefer the decorators over the context, but I guess we could have both. Either way, the memory map support has to exist first in C.

andrewleech, 2025-03-29:
I've thought about context managers as a clean syntax for times when python needs ram from a particular source, or aligned etc, that has my vote! 
There's slight complications in that it would probably need to be thread aware to ensure the hint passed to the gc by the context manager only affects the current thread, but otherwise should be fairly straightforward.

Gadgetoid, 2025-03-29:
> In python syntax, are decorators allowed on anything other than a function declaration?

Currently no, the conceit is that it could be so on MicroPython but then we never know if that might cause some future turmoil if CPython allows them on statements. A context manager is probably the safest bet.

> Either way, the memory map support has to exist first in C.

And, indeed, step one is to mark up RAM with the appropriate attributes!

agatti, 2025-03-30:
Passing buffers to ESP32S3's SIMD unit could also benefit from this - my DSP code was spending an awful amount of time to handle unaligned loads and stores, and it was faster (in every sense of the word) to move buffer management to the C side so I could have a properly aligned buffer to pass to SIMD code.