micropython/micropython #16646: UART not fast enough

quantrpeter, 2025-01-25:
### Port, board and/or hardware

esp32

### MicroPython version

Hi. Is the UART not fast enough? If I paste code into UART terminal, the code becoming crazy.

### Reproduction

past a code with over 250 characters.

### Expected behaviour

the code should be pasted correctly

### Observed behaviour

the code pasted into the terminal will become wrong sequence of characters.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-01-25:
Which exact ESP32C6 board are you using?

The UART does not have flow control and there is a finite input buffer, so it is possible to overrun that buffer.  This is a limitation of the board you are using.

quantrpeter, 2025-01-25:
Hi @dpgeorge  I am using this one https://www.waveshare.net/wiki/ESP32-C6-Pico

quantrpeter, 2025-01-27:
I have tried another board raspberry pi pico, same story, any hints please?

quantrpeter, 2025-01-28:
I have figured it out, I have written a tutorial for everyone https://peter.quantr.hk/2025/01/the-correct-way-to-send-code-to-micropython-via-uart-and-run/

closed 2025-01-28 by quantrpeter