micropython/micropython #16650: TLS errors on 5.4 with default MICROPY_GC_INITIAL_HEAP_SIZE

chaseadam, 2025-01-25:
### Port, board and/or hardware

esp32 port

### MicroPython version

```
➜  esp32 git:(master) git describe --dirty
v1.24.0-224-ga4ab84768
```

### Reproduction

Making single requests to 2 different TLS endpoints (which have small payloads).

```
➜  esp32 git:(master) ✗ idf.py --version
ESP-IDF v5.4
...
Chip is ESP32-D0WDQ6 (revision v0.0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
```

### Expected behaviour

IDF 5.3 was stable without change to MICROPY_GC_INITIAL_HEAP_SIZE

```
➜  esp32 git:(master) idf.py --version
ESP-IDF v5.3
```

### Observed behaviour

I was getting various intermittent lock up (including no ping responses via network to device) or OSError with TLS HTTP requests:
(-17040, 'MBEDTLS_ERR_RSA_PUBLIC_FAILED+MBEDTLS_ERR_MPI_ALLOC_FAILED')
(-29312, 'MBEDTLS_ERR_SSL_CONN_EOF')
[Errno 104] ECONNRESET

Workaround:  Adjusting MICROPY_GC_INITIAL_HEAP_SIZE to 52 (from 56) none of these errors triggered. Errors triggered if I had the value set to 54 as well.

### Additional Information

It is unstable on 5.3.2 as well, but I noted it is not in the "approved" IDF versions. I am not sure if the patch version matters, but because 5.2 and 5.2.2 are explicitly listed, it makes me think they are. Unfortunately, IDF does not have a 5.3.0, so I think micropython esp32 port just lists 5.3. May be worth indicating that supported patch versions will be explicitly listed.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-01-26:
> ### Reproduction
> 
> Making single requests to 2 different TLS endpoints (which have small payloads).

Can you please provide a Python code sample which triggers the error?

chaseadam, 2025-01-27:
it is a bit of a cobbled together mess, but this reproduces a lockup on a raw ESP32 module:
https://github.com/chaseadam/PlasticPlayerRedux/tree/micropython-issue-16650

```
➜  esp32 git:(master) git describe --dirty
v1.24.0-224-ga4ab84768
➜  esp32 git:(master) make BOARD=ESP32_GENERIC BOARD_VARIANT=OTA FROZEN_MANIFEST=${HOME}/git/PlasticPlayerRedux/manifest.py
➜  esp32 git:(master) ESPTOOL_PORT=/dev/ttyUSB0 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x1000 build-ESP32_GENERIC-OTA/bootloader/bootloader.bin 0x8000 build-ESP32_GENERIC-OTA/partition_table/partition-table.bin 0xd000 build-ESP32_GENERIC-OTA/ota_data_initial.bin 0x10000 build-ESP32_GENERIC-OTA/micropython.bin
esptool.py v4.8.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WDQ6 (revision v0.0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
```

I modified the flow a bit to remove requirement to interact with the device to trigger it.

All required libraries are in the manifest and frozen into the image.

Copy over boot.py and main.py.

Skip the WIFI setup by creating a `config.json` with contents:
```
{"ssid": "<SSID>", "psk": "<PSK>"}
```

`tcpdump` from the router where this traffic passes through showing the lock up from the network side:

```
15:36:02.726912 IP 192.168.5.172.63539 > 35.186.224.24.443: Flags [R.], seq 3244877987, ack 3397142613, win 32790, length 0
15:36:04.393912 IP 192.168.5.172.7406 > 192.168.5.1.53: 42226+ A? accounts.spotify.com. (38)
15:36:04.404452 IP 192.168.5.1.53 > 192.168.5.172.7406: 42226 2/0/0 CNAME edge-web.dual-gslb.spotify.com., A 35.186.224.24 (87)
15:36:04.411395 IP 192.168.5.172.63540 > 35.186.224.24.443: Flags [S], seq 3353017010, win 5760, options [mss 1440], length 0
15:36:04.422863 IP 35.186.224.24.443 > 192.168.5.172.63540: Flags [S.], seq 98823943, ack 3353017011, win 65535, options [mss 1412], length 0
15:36:04.426908 IP 192.168.5.172.63540 > 35.186.224.24.443: Flags [.], ack 1, win 5760, length 0
15:36:04.432859 IP 192.168.5.172.63540 > 35.186.224.24.443: Flags [P.], seq 1:246, ack 1, win 5760, length 245
15:36:04.444373 IP 35.186.224.24.443 > 192.168.5.172.63540: Flags [.], ack 246, win 65535, length 0
15:36:04.446623 IP 35.186.224.24.443 > 192.168.5.172.63540: Flags [.], seq 1:1413, ack 246, win 65535, length 1412
15:36:04.446715 IP 35.186.224.24.443 > 192.168.5.172.63540: Flags [P.], seq 1413:2049, ack 246, win 65535, length 636
15:36:04.446778 IP 35.186.224.24.443 > 192.168.5.172.63540: Flags [P.], seq 2049:3403, ack 246, win 65535, length 1354
15:36:04.449555 IP 192.168.5.172.63540 > 35.186.224.24.443: Flags [.], ack 2049, win 3712, length 0
15:36:04.453179 IP 192.168.5.172.63540 > 35.186.224.24.443: Flags [.], ack 3403, win 5413, length 0
15:36:09.526904 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:10.550885 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:11.574881 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:14.444075 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:15.446891 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:16.470881 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:17.506009 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:18.518899 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:19.542899 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:20.578287 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:21.590881 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:22.614881 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:23.650004 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:24.662881 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:25.686883 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:26.721987 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:27.734894 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:28.758879 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:29.793967 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:30.806884 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:31.830875 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:32.866039 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:33.878890 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:34.902889 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:35.938386 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:36.950884 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:36:37.974879 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
[... 71 lines cut ...]
15:39:43.204280 IP 192.168.5.172.65234 > 35.186.224.24.443: Flags [.], ack 2049, win 3712, length 0
15:39:43.207021 IP 192.168.5.172.65234 > 35.186.224.24.443: Flags [.], ack 2049, win 5760, length 0
15:39:43.403443 IP 192.168.5.172.65234 > 35.186.224.24.443: Flags [.], ack 3403, win 4406, length 0
15:39:43.614263 IP 192.168.5.172.65234 > 35.186.224.24.443: Flags [P.], seq 246:321, ack 3403, win 4406, length 75
[... cut ...]
15:39:43.891046 IP 192.168.5.172.5353 > 224.0.0.251.5353: 0*- [0q] 1/0/0 (Cache flush) A 192.168.5.172 (46)
15:39:43.923688 ARP, Request who-has 192.168.5.172 tell 192.168.5.172, length 46
15:39:44.566897 ARP, Request who-has 192.168.5.172 tell 192.168.5.1, length 28
15:39:44.657525 ARP, Reply 192.168.5.172 is-at 30:ae:a4:02:88:08, length 46
15:39:44.924522 ARP, Request who-has 192.168.5.172 tell 192.168.5.172, length 46
15:39:44.991056 IP 192.168.5.172.5353 > 224.0.0.251.5353: 0*- [0q] 1/0/0 (Cache flush) A 192.168.5.172 (46)
```