micropython/micropython #16684: unable to create genuine global variables throughout all imported modules

xuancong84, 2025-02-02:
### Description

The standard C-Python allows creation of genuine global variables across all imported modules in the following way:
```
import os
os.g_abc = "Hello world"
```
With this, inside any imported module, you can access `os.g_abc` after importing `os` module.

However, this is not allowed in Micropython, giving the following error:
```
AttributeError: 'module' object has no attribute 'g_abc'
```

FYI, this is the only way to create genuine global variables if your level of python profession believes that `globals()` can solve the problem. No. That is because `globals()` is only global w.r.t. the module that it is located in, not w.r.t. the global runtime. For example,
```
### moduleA.py

def test():
 globals()['g_abc'] = 'hello'

def mainA():
 print(g_abc)
```

```
### moduleB.py

import moduleA
moduleA.test()

moduleA.mainA()   # this works fine
print(globals()['g_abc']). # this will crash, because the global variable is only defined within moduleA context, not in moduleB or elsewhere
```

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

peterhinch, 2025-02-02:
I think this arises because the code is modifying a built-in module (os) whose code is ROM-based. The following works:
```py
# modulea .py

a = "Hello"
```
```py
# moduleb.py

import modulea
print(modulea.a)
```
```
MicroPython v1.24.0-preview.395.ge5eeaa7df.dirty on 2025-02-01; linux [GCC 11.4.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import moduleb
Hello
>>> 
```

xuancong84, 2025-02-03:
Thanks @peterhinch for your prompt reply! Of course I know this workaround, but this behavior is inconsistent with the standard Python implementation so that I have to create an additional module with an additional file for publishing global variables. (`import os` can be done anywhere, but `import myglobals` cannot: for any python file that imports `myglobals`, if it is moved to a different path, you have to change the import statement or add to `os.path` which will incur further complications.)

Moreover, sometimes I want to bind the STDIN/STDOUT before calling another module:
```
tmp_stdout = sys.stdout # save STDIO
sys.stdout = StringIO()
external_func() # call external module with redirected STDIO
sys.stdout = tmp_stdout # restore STDIO
```
Thus, the inability to modify builtin module objects will pose many hindrances to many tasks.

Technically, it is possible for MicroPython to achieve the standard implementation. We can create an overlay object in RAM (similar to the [overlay file-system](https://en.wikipedia.org/wiki/OverlayFS) principle) that overrides builtin objects. Whenever its member element is accessed:
- for read operation, the interpreter will first look at its RAM object, if not found then look at its ROM object.
- for write operation, the interpreter will write to its RAM object.
- for delete operation, the interpreter will assign a special symbol to its RAM object so that read operation will stop here and will not proceed to look at its ROM object.

dpgeorge, 2025-02-04:
It's possible to extend built-in modules in Python, and then you can modify their globals.  And some of the modules already have extensions available for them.  For example if you run:
```sh
$ mpremote mip install os
```
Then on your board you can now do:
```py
>>> import os
>>> os.foo = 123
```
Thus, you can modify built-in modules (and in general this is MicroPython's philosophy, than there's usually a way to optionally extend things to more closely match CPython).

xuancong84, 2025-02-04:
Thanks to @dpgeorge , your solution is a working one. Does it incur more RAM consumption? Because built-in modules are stored in ROM, importing them does not consume more RAM (except for the linkage pointer in global namespace).

dpgeorge, 2025-02-04:
> Does it incur more RAM consumption?

Yes it does.  The `os` module's globals dict is essentially copied across to RAM, and that's why it can now be modified.

But this is the price to pay for the ability to modify a built-in module.  However it's implemented, the ability to modify a built-in module will cost RAM at some point.