micropython/micropython #16692: Build fails with GCC 15 due to  -Wunterminated-string-initialization

stratakis, 2025-02-04:
### Port, board and/or hardware

x86_64

### MicroPython version

1.24.1

### Reproduction

When compiling micropython on Fedora with the latest GCC 15, a new warning is now enabled by default through -Wextra:  -Wunterminated-string-initialization.

This makes the compilation fail for the arrays at:
https://github.com/micropython/micropython/blob/master/py/emitinlinethumb.c#L152
https://github.com/micropython/micropython/blob/v1.24.1/py/emitinlinethumb.c#L370
https://github.com/micropython/micropython/blob/v1.24.1/py/emitinlinethumb.c#L391
https://github.com/micropython/micropython/blob/v1.24.1/py/emitinlinethumb.c#L430 
https://github.com/micropython/micropython/blob/master/py/emitinlinextensa.c#L121

### Expected behaviour

Compilation passes.

### Observed behaviour

These errors are observed for each array element.

<snippet>

`../py/emitinlinextensa.c:122:9: error: initializer-string for array of ‘unsigned char’ is too long [-Werror=unterminated-string-initialization]
  122 |     {0, "a0\0"}`

`../py/emitinlinethumb.c:153:9: error: initializer-string for array of ‘unsigned char’ is too long [-Werror=unterminated-string-initialization]
  153 |     {0, "r0\0"}`

`../py/emitinlinethumb.c:371:24: error: initializer-string for array of ‘unsigned char’ is too long [-Werror=unterminated-string-initialization]
  371 |     { ASM_THUMB_CC_EQ, "eq" }`

`../py/emitinlinethumb.c:394:34: error: initializer-string for array of ‘char’ is too long [-Werror=unterminated-string-initialization]
  394 |     { X(ASM_THUMB_FORMAT_4_LSR), "lsr" },`

`../py/emitinlinethumb.c:431:13: error: initializer-string for array of ‘char’ is too long [-Werror=unterminated-string-initialization]
  431 |     { 0x30, "add" },`
<snippet>

Full [build.log](https://github.com/user-attachments/files/18650365/build.log) is attached.

### Additional Information

Similar issues from other projects:
https://github.com/Mbed-TLS/mbedtls/issues/9944
https://github.com/besser82/libxcrypt/issues/194

### Code of Conduct

Yes, I agree

projectgus, 2025-02-04:
Thanks @stratakis for the heads-up.

It looks like there's not exactly a bug or UB in the MicroPython usage of this (the extended discussion of the latter in the mbedTLS issue was helpful), but I can see why the warning was added.

I think our options will be either to disable that warning (difficult as it's not in older GCC), or to rework all these initialisers to use inline array syntax, i.e. `"a1\0"` becomes `{'a','1'}` .

If you or anyone reading this is feeling generous then a PR that made those array changes would probably be merged. Otherwise we'll most likely wait for the GCC 15 release to confirm this warning still triggers on these usages.[*]

[*] This may seem like wishful thinking on my part, but it does seem to me there's an argument for winding this warning back so it only applies if you're initialising an array of `char`, not an array of a different byte-size type. i.e. https://gcc.godbolt.org/z/5PPr9df6e - even though the latter can still be passed as-is to a string API, I suppose.

projectgus, 2025-02-04:
> GCC 15 release to confirm this warning still triggers on these usages.

There's an open request to disable this warning for fields marked `__attribute__((nonstring))`: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=117178

Unfortunately this (a) has to be applied per-field and (b) isn't implemented yet, and unclear if it'll implemented for the release. But 🤞

jwakely, 2025-02-05:
Using `"a0\0"` to initialize a `char[3]` is strange and completely unnecessary. `"a0"` already has a terminating NUL and is the right length, and of course includes the NUL in the initialized array. Changing that wouldn't help with the cases like `"r11"` though as those will still warn (and won't include any NUL in the array). You could use `{'r', '1', '1'}` for those cases.

Or just disable the warnings.

agatti, 2025-02-07:
@projectgus For the Xtensa register lookups I've got https://github.com/agatti/micropython/commit/7e86d101a5fc594445754b805480a757a3cfdb1f, that's part of another PR I'm working on (proper Xtensa inline assembler support) - if needed I can extract it as its own PR.  It also makes the code smaller by 40 bytes or so ;)

projectgus, 2025-02-10:
> GCC always ignores unknown `-Wno-foo` options, precisely so that you **don't** have that difficulty.

Thanks @jwakely. I've relied on this behaviour in the past and clean forgot about it last week!

@dpgeorge Do you have a preference between disabling this warning and refactoring to use array initializer syntax?

projectgus, 2025-02-10:
> [@projectgus](https://github.com/projectgus) For the Xtensa register lookups I've got [agatti@7e86d10](https://github.com/agatti/micropython/commit/7e86d101a5fc594445754b805480a757a3cfdb1f), that's part of another PR I'm working on (proper Xtensa inline assembler support) - if needed I can extract it as its own PR. It also makes the code smaller by 40 bytes or so ;)

I had been wondering whether these are worth interning as qstrs, thanks for pointing this out @agatti. I guess this is a third optional approach, ideally we'd apply it for all of the native emitters but we could always start with xtensa.

agatti, 2025-02-10:
> ideally we'd apply it for all of the native emitters but we could always start with xtensa

For what it's worth, RV32 already does - although it needs #16524 to use 16 bits wide qstr indices rather than 32 bits ones as it is right now.

agatti, 2025-05-01:
Is anybody working on a PR to fix the remaining issues?  Arch Linux pushed GCC 15.1 as the current compiler version and I'm stuck even for embedded targets, as `mpy-cross` not building anymore.

If no-one is on this I don't mind submitting one myself, but I don't want to duplicate efforts if this is already worked on.  I didn't find any open PR so no idea if this is being fixed behind the scenes.

dpgeorge, 2025-05-01:
See #17215.  That should hopefully fix the remaining issues.

(I tried doing the QSTR thing but it increased size by a lot....)

agatti, 2025-05-01:
I'll temporarily merge that into my tree, thanks!

Regarding QSTR inlining, just rewriting the register lookup code should still save some bytes.

Maybe it's out of scope of this specific issue, but there's another GCC 15.1 feature to watch out for (from https://gcc.gnu.org/gcc-15/changes.html):

> {0} initializer in C or C++ for unions no longer guarantees clearing of the whole union (except for static storage duration initialization), it just initializes the first union member to zero. If initialization of the whole union including padding bits is desirable, use {} (valid in C23 or C++) or use -fzero-init-padding-bits=unions option to restore old GCC behavior.

There are a few instances in both `py/` and `ports/` - wonder if that's something needing attention or not.

closed 2025-05-09 by projectgus

stratakis, 2025-05-09:
Thanks for the fixes!

PR #17269 (2025-05-09): lib,py,extmod: Fixes for GCC 15.1 unterminated string literal warning
### Summary

This PR is a collection of commits to fix MicroPython building under GCC 15.1 (host x86-64 toolchain, follow-up fixes may be necessary in the future for cross-toolchains).

* Fixes the new -Wunterminated-string-literal warning by using array initialisers instead of string literals. This is an alternative approach to #17215. Closes #16692
* ~~Disable the same warning in LFS1, as that upstream is no longer maintained. (Thanks @DvdGiessen for noting this in https://github.com/micropython/micropython/pull/17215#issu [... cut]