micropython/micropython #16701: io.IOBase bug

lbuque, 2025-02-05:
I am testing the io.IOBase class.

I am using this firmware https://micropython.org/resources/firmware/ESP32_GENERIC_S3-20241129-v1.24.1.bin and running the following code:

```python
import machine

uart = machine.UART(
    2,
    baudrate=115200,
    bits=8,
    parity=None,
    stop=1,
    tx=17,
    rx=18,
    txbuf=256,
    rxbuf=256,
    timeout=0,
    timeout_char=0,
    invert=0,
    flow=0,
)

import time
uart.write("+++ATH\r\n")
time.sleep(0.1)
uart.write("AT+CGDCONT=1,\"IP\",\"cmnet\"\r\n")
time.sleep(0.1)
uart.write("ATD*99#\r\n")
time.sleep(0.1)
uart.read()

import io

class VirUart(io.IOBase):

    _MP_STREAM_POLL = const(3)
    _MP_STREAM_POLL_RD = const(0x0001)

    def __init__(self, uart):
        self.uart = uart

    def readinto(self, buf):
        return self.uart.readinto(buf)

    def write(self, buf):
        return self.uart.write(buf)

    def ioctl(self, op, arg):
        if op == self._MP_STREAM_POLL:
            if self.uart.any():
                return self._MP_STREAM_POLL_RD
        return 0

import network
ppp = network.PPP(VirUart(uart))
ppp.active(True)
ppp.connect()
```

Unfortunately, I didn't get it to work correctly and got the following exception:

```
8
27
9
b'\x00ATH\r\r\nOK\r\n\r\nPB DONE\r\nAT+CGDCONT=1,"IP","cmnet"\r\r\nOK\r\nATD*99#\r'
True

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : v1.24.1 on 2024-11-29
IDF version : v5.2.2
Machine     : Generic ESP32S3 module with ESP32S3

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42039766  PS      : 0x00060530  A0      : 0x820108a6  A1      : 0x3fcbba20  
A2      : 0x3c1815b0  A3      : 0x3c1815f0  A4      : 0x0000000f  A5      : 0x00000002  
A6      : 0x00000001  A7      : 0x00000200  A8      : 0x82039766  A9      : 0x3fcbba00  
A10     : 0x00000000  A11     : 0x3c1815f0  A12     : 0x3c1815b0  A13     : 0x3fcbbad0  
A14     : 0x00000001  A15     : 0x00000002  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x42039763:0x3fcbba20 0x420108a3:0x3fcbba50 0x42017e31:0x3fcbba80 0x42017ef9:0x3fcbbaa0 0x42041ed1:0x3fcbbac0 0x42041f05:0x3fcbbb00 0x42116bcd:0x3fcbbb20 0x4202e79d:0x3fcbbb50 0x420a0964:0x3fcbbb80 0x420a0ad4:0x3fcbbba0 0x421199f5:0x3fcbbbd0 0x420a98ef:0x3fcbbbf0 0x420a994b:0x3fcbbc20 0x420a7a2b:0x3fcbbc40 0x420a0435:0x3fcbbc60 0x420a084d:0x3fcbbc80 0x420a02a5:0x3fcbbca0 0x420a0730:0x3fcbbcc0 0x42091da1:0x3fcbbce0
```

```
$ xtensa-esp32s3-elf-addr2line -e ~/Downloads/ESP32_GENERIC_S3-20241129-v1.24.1.elf Backtrace: 0x42039763:0x3fcbba20 0x420108a3:0x3fcbba50 0x42017e31:0x3fcbba80 0x42017ef9:0x3fcbbaa0 0x42041ed1:0x3fcbbac0 0x42041f05:0x3fcbbb00 0x42116bcd:0x3fcbbb20 0x4202e79d:0x3fcbbb50 0x420a0964:0x3fcbbb80 0x420a0ad4:0x3fcbbba0 0x421199f5:0x3fcbbbd0 0x420a98ef:0x3fcbbbf0 0x420a994b:0x3fcbbc20 0x420a7a2b:0x3fcbbc40 0x420a0435:0x3fcbbc60 0x420a084d:0x3fcbbc80 0x420a02a5:0x3fcbbca0 0x420a0730:0x3fcbbcc0 0x42091da1:0x3fcbbce0
??:0
/home/micropython/micropython-autobuild/py/cstack.c:46
/home/micropython/micropython-autobuild/py/objfun.c:230
/home/micropython/micropython-autobuild/py/runtime.c:717
/home/micropython/micropython-autobuild/py/runtime.c:733
/home/micropython/micropython-autobuild/py/modio.c:59
/home/micropython/micropython-autobuild/py/modio.c:77
/home/micropython/micropython-autobuild/py/stream.c:60
/home/micropython/micropython-autobuild/ports/esp32/network_ppp.c:104
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/netif/ppp/pppos.c:878
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/netif/ppp/pppos.c:241
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/netif/ppp/ppp.c:996
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/netif/ppp/fsm.c:757
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/netif/ppp/fsm.c:102
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/netif/ppp/lcp.c:475
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/netif/ppp/ppp.c:750
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/netif/ppp/pppos.c:340
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/netif/ppp/ppp.c:464
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/netif/ppp/pppapi.c:277
/home/micropython/esp-idf-v5.0/components/lwip/lwip/src/api/tcpip.c:166
```

What am I doing wrong?

_Originally posted by @lbuque in https://github.com/micropython/micropython/discussions/16699_

lbuque, 2025-02-05:
Running the micropython/examples/bluetooth/ble_uart_repl.py script gives an exception. Could they be the same problem?

[image]

jonnor, 2025-03-03:
One is a crash and another is an exception, it does not seem very likely to be the same issue.

Are you able to further reduce your code for reproducing this issue? Like, can it reproduce with IOBase without the PPP driver? That would considerably help in finding the problem.

dpgeorge, 2025-03-05:
Unfortunately you can't use a user stream with esp32's `network.PPP` class.  That's because the PPP driver runs a task to process the PPP data, and that task is independent to the MicroPython task, and in  particular cannot run Python code (eg it can't run your `VirUart.readinto` method).

So, for now you'll just need to pass the `machine.UART` instance directly to the `network.PPP` constructor.