micropython/micropython #16709: Delayed PIO Assambly with custom variables

HighImp, 2025-02-05:
### Description

Hi,

I would like to start a pull request for a alternative of the decorator: ```def asm_pio(**kw):``` that can be called afterward.
This change would allow all "constants" to be modified during runtime in Python, significantly improving flexibility.

We use PIOs in our project for various situations. See my [Reddit post](https://www.reddit.com/r/MicroPythonDev/comments/1i6utf6/comment/m8w57ek/) for a specific use case.
The discussed solution requires two state machines instead of one.

The adaptation is ready and working—see also #7903, where I posted my first proposal for the new behavior and the [Branch](https://github.com/HighImp/micropython/tree/topic/add_delay_assambly) if you are interested.

Best regards!

### Code Size

I am not sure how much code will be used, but the adaptation in the branch takes 63 additional lines. During operation, the fact that the function will not collapse instantly into a tuple of bytes may require more RAM than the original approach. However, I think it is worth it, considering the amount of saved space in the PIO modules. I would love to hear your thoughts!

### Implementation

I intend to implement this feature and would submit a Pull Request if desirable

### Code of Conduct

Yes, I agree

Gadgetoid, 2025-02-21:
This is interesting, and I'd probably use a feature like this, but to play devil's advocate I have a couple of concerns:

1. I'm not sure maintaining two separate functions with extremely similar behaviour is a reasonable approach.
2. By polluting the function namespace with `gl.update(variables)` you're missing the chance to be explicit about what's parametisable
3. The PIO compiler is pretty sophisticated if you think about it, and limiting variables to 0-31 ints is unnecessary

Consider this horrible hack of blink_1hz.py (the timings are off on RP2350 🤔 which throws an exception if the PIO freq is set to 2000Hz... but that's uh, another issue):

```python
# Example using PIO to blink an LED and raise an IRQ at 1Hz.
# Note: this does not work on Pico W because it uses Pin(25) for LED output.

# ruff: noqa: F821 - @asm_pio decorator adds names to function scope

import time
from machine import Pin
import param_rp2 as rp2

DELAY_TIME = 1

@rp2.asm_pio(set_init=rp2.PIO.OUT_LOW)
def blink_fixed():
    # fmt: off
    # Cycles: 1 + 1 + 6 + 32 * (30 + 1) = 1000
    irq(rel(0))
    set(pins, 1)
    set(x, 31)                  [5]
    label("delay_high")
    nop()                       [29]
    jmp(x_dec, "delay_high")

    # Cycles: 1 + 1 + 6 + 32 * (30 + 1) = 1000
    nop()
    set(pins, 0)
    set(x, 31)                  [5]
    label("delay_low")
    nop()                       [29]
    jmp(x_dec, "delay_low")
    # fmt: on

def blink_program(delay_time, nops):
    # fmt: off
    # Cycles: 1 + 1 + 6 + 32 * (30 + 1) = 1000
    irq(rel(0))
    set(pins, 1)
    set(x, 31)                  [5]
    label("delay_high")
    nop()                       [delay_time]
    for _ in range(nops - 1):
        nop()                   [delay_time - 1]
    jmp(x_dec, "delay_high")

    # Cycles: 1 + 1 + 6 + 32 * (30 + 1) = 1000
    nop()
    set(pins, 0)
    set(x, 31)                  [5]
    label("delay_low")
    nop()                       [delay_time]
    for _ in range(nops - 1):
        nop()                   [delay_time - 1]
    jmp(x_dec, "delay_low")
    # fmt: on
    
blink_1hz = rp2.asm_pio(program=blink_program, set_init=rp2.PIO.OUT_LOW, args={"delay_time": 29, "nops": 2})
blink_2hz = rp2.asm_pio(program=blink_program, set_init=rp2.PIO.OUT_LOW, args={"delay_time": 29, "nops": 4})
blink_normal = rp2.asm_pio(program=blink_program, set_init=rp2.PIO.OUT_LOW, args={"delay_time": 29, "nops": 1})

assert blink_1hz != blink_2hz
assert blink_normal == blink_fixed

# Create the StateMachine with the blink_1hz program, outputting on Pin(25).
sm = rp2.StateMachine(0, blink_2hz, freq=4000, set_base=Pin(25))

# Set the IRQ handler to print the millisecond timestamp.
#sm.irq(lambda p: print(time.ticks_ms()))

# Start the StateMachine.
sm.active(1)

```

Once you parameterise PIO, and even before, you can get *really* fancy with it. Conditionally inserting or repeating instructions being particularly spicy.

This entire feature is two extra lines of code and 6 total line changes:

```diff
diff --git a/ports/rp2/modules/rp2.py b/ports/rp2/modules/rp2.py
index 6068926036..c2832b5ba0 100644
--- a/ports/rp2/modules/rp2.py
+++ b/ports/rp2/modules/rp2.py
@@ -241,10 +241,13 @@ _pio_instructions = (
 
 
 def asm_pio(**kw):
+    fn = kw.pop("program") if "program" in kw else None
+    args = kw.pop("args") if "args" in kw else {}
+
     emit = PIOASMEmit(**kw)
 
     def dec(f):
-        nonlocal emit
+        nonlocal emit, args
 
         gl = f.__globals__
         old_gl = gl.copy()
@@ -257,17 +260,17 @@ def asm_pio(**kw):
             gl[name] = getattr(emit, name)
 
         emit.start_pass(0)
-        f()
+        f(**args)
 
         emit.start_pass(1)
-        f()
+        f(**args)
 
         gl.clear()
         gl.update(old_gl)
 
         return emit.prog
 
-    return dec
+    return dec(fn) if fn else dec
 
 
 # sideset_count is inclusive of enable bit
```

Gadgetoid, 2025-02-21:
(yes you can use for loops in PIO asm, this is gloriously cursed and deliciously wonderful)

(if any parameters exceed the 0-31 or otherwise a limit of a single PIO instruction, then that instruction can presumably raise a `ValueError` just as it would if a user put 9999 in as an immediate constant value:

```
PIOASMError: delay too large
```

HighImp, 2025-02-21:
Looks great and much easier to use. I am not sure if I would set the maximum higher than 31, but you are right—if I use the PIO assembler, especially in this way, I need to know what happens.

HighImp, 2025-02-21:
So... do you create the pr or should I do it? I would love to assist if there is something to add, may adapt the documentation?

Gadgetoid, 2025-02-21:
Have at it! I need another PR to babysit like I need a headache! 🤣 just got a bit nerd sniped here and learned some things.

Gadgetoid, 2025-02-21:
Don’t take my code as gospel though, I’m a bad programmer who did the bare minimum of testing 🫠

mendenm, 2025-02-27:
I use this idiom to handle compile-time PIO variables.  
```
def neo_prog_side(T1=6, T2=7, T3=7):
    """define a pio program with the appropriate delays for the intended sm clock speed"""
    
    def func():   
        label("loop")
        out(x, 1).side(0).delay(T3-1)            # x <= left-most 1 bit of osr
        jmp(not_x, "zero").side(1).delay(T1-1)
        jmp("loop").side(1).delay(T2-1)
        label("zero")
        nop().side(0).delay(T2-1)       
        wrap()
    return func

class neopixel:
    pio32 = rp2.asm_pio(
        sideset_init=[rp2.PIO.OUT_LOW],
        out_shiftdir=rp2.PIO.SHIFT_LEFT,
        autopull = True,
        pull_thresh = 32
     ) ( neo_prog_side(3,3,3) ) #override these to change clock speeds, e.g.
```
which generates a version of the PIO function with 3 waits instead of 6 or 7.

Gadgetoid, 2025-02-27:
@mendenm that's clever, thank you! This feels a lot cleaner at the calling phase, though it's syntactically a little confusing. The one thing I really, really don't like about the approach we've bashed together here is passing args like this:

```
args={"delay_time": 29, "nops": 2}
```

It's way more explicit and easy to figure out what's going on at a glance than your clever workaround, but it's also really wordy (and requires changes whereas yours is magic 😆 ).

Since you've got a horse in this race, how does this PR look to you? https://github.com/micropython/micropython/pull/16800

The other benefit of your approach is that you've got an outer context (I'm surprised the `neo_prog_side` scope is available to `func` actually) where you could, for example, calculate the values of T1, T2 and T3 based upon a single desired frequency.

Anyway... I'm not totally sold on either approach, mostly because I never even considered this as a possibility (too much chucking PIO code into C 😬 modules).

mendenm, 2025-03-01:
@Gadgetoid:  I looked at your PR.  It is sufficiently lightweight that it is hard to argue with it as one perfectly good way to accomplish this.

Your question/comment about the availability of the `neo_prog_side` scode inside `func` comes down to standard python scoping.  You really only need the `nonlocal` declarations to make sure python examines outer scopes before it gets to the global scope when you’re are trying to modify a variable that appears in an outer scope that isn’t global.  Read access is (as far as I know) unaffected by such declarations, since python always starts working from the inside out.