micropython/micropython #16713: Serial com and external power supply

eleroy, 2025-02-06:
### Port, board and/or hardware

rp2, raspberry pico

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Raspberry Pi Pico with RP2040

### Reproduction

1. Start a code with serial printing such as:

```
import time

while True:
    print(time.time())
    time.sleep(0.1)    
```

2. Connect external power supply (+5V or 3V3)
3. Disconnect usb
4. Wait ~4 s
5. Reconnect usb

### Expected behaviour

When you reconnect USB, the board should be normally detected as CDC Serial device and printing.

### Observed behaviour

When you reconnect USB the board is not detected anymore and it seems the program has been interrupted.

### Additional Information

I suspect something like a buffer overflow on the serial output hardware is happening.

If the usb is reconnected quickly (in 1 second or so), the board is well detected.

### Code of Conduct

Yes, I agree

Josverl, 2025-02-09:
Have you tried blinking a led to see the state of the script independent of an established connection from the host?

Also depending on the details of the USB CDC/UART connection, it's common that new usb devices are detected by the host based on a logical low on the USB bus. If that's high the device may not be detected , nor initialized on the host.

On windows I check with a `usbtreeview.exe`, on linux you need a handful of cmdline tools that I can seem to remember, `lsusb` is one of them

eleroy, 2025-02-10:
Hello @Josverl,

Thank you for your answer, that is indeed a good idea to make a led blink to check the status of the program. It helped me to understand the behavior.

Here is the code, please note that Micropython is not throwing any exception (so the second part of the code is useless).

```python
import time
import machine
pin = machine.Pin(25, machine.Pin.OUT)
try:
    while True:
        print(time.time())
        pin.toggle()
        time.sleep(0.25)    
except Exception as e:
    with open("dump_file.txt", "a") as fh:
        fh.write(f"{time.time()},")
        fh.write('"')
        sys.print_exception(e, fh)  # type: ignore
        fh.write('"\n')
    sys.print_exception(e)  # type: ignore
    sys.exit()
```

I can confirm that under certain conditions, disconnecting the usb serial **while a com port is open** and the controller isprinting
causes the program to crash after 5 s or so (it stops the led blinking). 

Steps to reproduce:

1. Place a code with a regular printing statement e.g. `print(time.time())`
2. Open the serial port (I tested under windows)
3. Power up external power supply (It can probably be done before 2.)
4. Disconnect the usb port **without properly closing** the com port
5. Wait for 5 seconds (approximatively) the program will interrupt.

If you reconnect before 5 seconds even if you don't open the serial port, the program will work as expected.

It must have something to do with serial port deinit which is not correctly done when usb is unplugged without closing serial port.

I'm not sure if it is micropython related or tinyusb related.

eleroy, 2025-02-10:
Just checked with 1.23 and it does work with this version.

agatti, 2025-02-14:
> I suspect something like a buffer overflow on the serial output hardware is happening.

I was able to reproduce this with a SWD debugger attached.  TinyUSB is deadlocked waiting on a mutex when attempting to write to the USB during when the regular USB connection setup is happening.

MyRaspberry, 2025-02-23:
using 
PICO W RP2040
MicroPython 1.23 ( newer versions not work using TLS )
and i use
MQTT connect ( umqtt.simple.py 1.15.0 )
it works as long i start THONNY main.py RUN

production RUN ( like from power supply ) hangs at that mqttclient.connect() line

confirmed with BLINK