micropython/micropython #16732: Reinitializing network.LAN with W5500 Causes Connection Loss Until Hard Reset

hosseinghaheri, 2025-02-10:
### Port, board and/or hardware

esp32 port

### MicroPython version

All Versions
MicroPython v1.25.0-preview.283.g11c9656fa on 2025-02-10; Generic ESP32 module with SPIRAM with ESP32

### Reproduction

# Issue Description:

When running the following MicroPython script on an ESP32 with a W5500 Ethernet module, the connection works correctly on the first run after powering up the board. However, if the script is executed again without a full power cycle or machine.reset(), the device loses network connectivity (cannot be pinged).

1. Power on the ESP32 with the W5500 module connected.
2. Run the following MicroPython script:

```python
import network
import time
from machine import Pin, SPI, reset

rst = Pin(15, Pin.OUT)
rst.value(0)
time.sleep(0.1)
rst.value(1)
time.sleep(0.5)

spi_nic = SPI(1, baudrate=26000000, sck=Pin(18), mosi=Pin(21), miso=Pin(19))

nic = network.LAN(phy_type=network.PHY_W5500, spi=spi_nic, phy_addr=1, cs=Pin(5), int=Pin(4))

config = ('192.168.1.120', '255.255.255.0', '192.168.1.1', '8.8.8.8')
nic.ifconfig(config)
nic.config(mac=b'\x02\xAB\xCD\xEF\x01\x02')

nic.active(True)
time.sleep(0.5)

while not nic.isconnected():
    print(".", end="")
    time.sleep(0.25)

print("\r\nip=", nic.ifconfig())

```

1. Confirm that the device is reachable via ping.
2. Run the script again without resetting the board.
3. Observe that the device no longer responds to ping, and the network connection is lost.
4. If machine.reset() is executed before re-running the script, the connection works fine again.

### Expected behaviour

Re-running the script should properly reset the W5500 and establish a new network connection without requiring a full board reset.

### Observed behaviour

After executing the script multiple times without a full reset, the ESP32 loses network connectivity. The W5500 module does not seem to reinitialize correctly unless a full machine.reset() is performed.

### Additional Information

# Workarounds Tried:

Explicitly deactivating nic before reinitializing:

```python
try:
    nic.active(False)
    del nic
except:
    pass
```

→ Did not solve the issue.
Resetting the W5500 before reinitialization using rst pin:
→ No effect; issue persists.
Performing machine.reset() before re-running the script:
→ Works, but not an ideal solution

### Code of Conduct

Yes, I agree

projectgus, 2025-02-18:
Thanks for the very detailed bug report.

I assume pin 15 is connected to the reset pin on the W5500 chip? If so, the root cause is that the `network.LAN` configuration stays active over a soft reset (i.e. after soft reset the network is still up), but toggling the reset pin clears the W5500 state and leaves the chip and the driver out of sync.

The best fix here is to remove the manual reset and pass the `reset` pin to the `network.LAN` constructor , like this:

```py
nic = network.LAN(phy_type=network.PHY_W5500, spi=spi_nic, phy_addr=1, cs=Pin(5), int=Pin(4), reset=Pin(15))
```

The LAN driver knows internally if the chip is already configured, and therefore can tell whether to toggle the reset line or not.

There currently isn't a way to manually send the network all the way back to "unconfigured" state from Python code, aside from a hard reset. i.e. `nic.active(False)` only temporarily disables it, because `nic.active(True)` is expected to resume it. 

This limitation has the upside that it is possible to keep the network up seamlessly over a soft reset. For example, for a faster soft reset you could write the code like this:

```py

nic = network.LAN(phy_type=network.PHY_W5500, spi=spi_nic, phy_addr=1, cs=Pin(5), int=Pin(4), reset=Pin(15))

if not nic.isconnected():
    # do initialisation steps here
```

hosseinghaheri, 2025-02-19:
@projectgus 
I added the reset pin in the network.LAN constructor, and the issue was resolved.
Thank you for your guidance!

projectgus, 2025-02-19:
@hosseinghaheri Great news!

@yfyungit If the suggestion above doesn't resolve your issue then please open a new issue with more details about how to reproduce.

closed 2025-02-19 by projectgus