micropython/micropython #16740: RP2 W fails to receive UDP datagrams

ryniker, 2025-02-12:
### Port, board and/or hardware

Pico W on Raspberry Pi Pico W

### MicroPython version

MicroPython v1.25.0-preview.295.g30acb16ad on 2025-02-11; Raspberry Pi Pico W with RP2040
  

### Reproduction

Simple receive program for the Pico W to run after
initializing the wireless hardware:

```
import socket
server_default_port = 50000
soc = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
soc.bind(('', server_default_port))
l, sender_address = soc.recvfrom(128)
print('l =', l)
```

This never returns from the call to recvfrom.
It works fine using an Espressif ESP-WROOM-32 board.

Simple test client:
```
import socket
server_default_port = 50000
IP_address = '192.168.100.91'
server_address = (IP_address, server_default_port)
soc = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
x = b'#0 D256\n'
n = soc.sendto(x, server_address)
print('sent', n, 'bytes', 'to', server_address)
```

### Expected behaviour

Should have received the transmitted message.

### Observed behaviour

No message received.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

ryniker, 2025-02-13:
Good explanation.  Not a code problem.  Perhaps a minor documentation issue since ports behave differently.

closed 2025-02-13 by ryniker

dpgeorge, 2025-02-13:
Might be good to make `socket.bind(("", port))` default to IPv4, but I don't know how to do that.

sosi-deadeye, 2025-02-13:
By default, `socket.socket()` uses IPv4: https://docs.python.org/3.13/library/socket.html#socket.socket
This function should fail, if the host does not have an IPv4 address. I've never tried it, because I never run an IPv6-Only network :-/

It would be handy, if you add create_server like the CPython implementation: https://docs.python.org/3.13/library/socket.html#socket.create_server
This could create a socket which supports dual stack mode.