micropython/micropython #16759: Asyncio task scheduling issue when using run_until_completed

yoctopuce, 2025-02-15:
### Port, board and/or hardware

any port with asyncio support

### MicroPython version

micropython-1.25.0 running on win32 [MSC v.1916 64 bit]

### Reproduction

The issue is linked to the fact that `run_until_complete` does not complete the processing of the _awaited_ task before returning to hte caller. If other pending tasks are waiting it, they will be lost forever.

The code below will reproduce the issue.
- When the tasks are await-ed form an async function, everything works as expected (function `async_test`)
- When the tasks are started from a sync function using run_until_complete, scheduling is broken  (function `sync_test`)
This issue is specific to micropython asyncio implementation of `run_until_complete` and does not occur in CPython.

```py
import sys, asyncio, random

def get_event_loop():
    if sys.implementation.name == "micropython":
        return asyncio.get_event_loop()
    else:
        try:
            eventloop = asyncio.get_running_loop()
        except RuntimeError:
            eventloop = asyncio.new_event_loop()
        return eventloop

class Worker:
    def __init__(self):
        self._eventLoop = None
        self._tasks = []

    def launchTask(self, asyncJob):
        if self._eventLoop is None:
            self._eventLoop = get_event_loop()
        return self._eventLoop.create_task(asyncJob)

    async def job(self, prerequisite, taskName):
        if prerequisite:
            await prerequisite
        await asyncio.sleep(2)
        print(taskName, "work completed")

    def planTasks(self):
        self._tasks.append(self.launchTask(self.job(None, 'task0')))
        self._tasks.append(self.launchTask(self.job(self._tasks[0], 'task1')))
        self._tasks.append(self.launchTask(self.job(self._tasks[1], 'task2')))

    async def waitForTask(self, taskIdx):
        return await self._tasks[taskIdx]

    def syncWaitForTask(self, taskIdx):
        return self._eventLoop.run_until_complete(self._tasks[taskIdx])

async def async_test():
    print("--- Running the async test")
    worker = Worker()
    worker.planTasks()
    await worker.waitForTask(0)
    print("-> task0 done")
    await worker.waitForTask(2)
    print("-> task2 done")

def sync_test():
    print("--- Running the sync test")
    worker = Worker()
    worker.planTasks()
    worker.syncWaitForTask(0)
    print("-> task0 done")
    worker.syncWaitForTask(2)
    print("-> task2 done")

asyncio.run(async_test())
sync_test()
```

### Expected behaviour

```
--- Running the async test
task0 work completed
-> task0 done
task1 work completed
task2 work completed
-> task2 done
--- Running the sync test
task0 work completed
-> task0 done
task1 work completed
task2 work completed
-> task2 done
```

### Observed behaviour

```
--- Running the async test
task0 work completed
-> task0 done
task1 work completed
task2 work completed
-> task2 done
--- Running the sync test
task0 work completed
-> task0 done
-> task2 done
```

### Additional Information

I will submit a pull request with a suggested fix

### Code of Conduct

Yes, I agree

peterhinch, 2025-02-15:
This is one of three issues whereby `asyncio` prematurely terminates. I wonder if there is a common cause? See https://github.com/micropython/micropython/issues/16569 and https://github.com/micropython/micropython/issues/16318.

yoctopuce, 2025-02-17:
The common _cause_ is that the scheduler exits when both the `_task_queue` and the `_io_queue` are empty, without checking if the `main_task` is really done or if it could still be in a different queue - typically an `Event` queue. In that sense, keeping track of task completion by always updating `task.data` and check for `main_task` completion after testing if `_io_queue.map` is empty before returning does solve the two other pending issues.

However what is different in that specific issue is that in order to allow `task1` and `task2` to start, we need to also make sure to push back in `_task_queue` the tasks that depend on `task0` before returning. I have tried several variants of fix for that, but for now my various attempts always affect one way or another some test cases, so I am still working on it...

dpgeorge, 2025-05-07:
Fixed by bdb7e036d2400c25beeb2b4252ff51c5255c4020

closed 2025-05-07 by dpgeorge