micropython/micropython #16771: ESP32-S3-WROOM-N8R2 WiFi Wont Connect

PsuFan, 2025-02-17:
### Port, board and/or hardware

ESP32-S3-WROOM-N8R2

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Generic ESP32S3 module with ESP32S3

### Reproduction

from network import WLAN
sta = WLAN(0)
sta.active(1)
sta.connect('net', 'pass')
sta.ifconfig()

### Expected behaviour

('192.168.1.100', '192.168.1.1', '0.0.0.0', '0.0.0.0')

### Observed behaviour

I (47019) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (47019) wifi:state: init -> auth (b0)
I (51319) wifi:state: auth -> init (200)
I (51319) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (51319) wifi: STA_DISCONNECTED, reason:2:
I (53729) wifi: STA_DISCONNECTED, reason:205:

### Additional Information

This a dual USB C board, I saw closed issues with this, txpower=17 did not work, tried both usb ports, no change. Tried WPA and no security on AP, no change. Other ESP32 and ESP32-S3 boards connect to AP fine. Almost decided to use ESP32 for project but some dumb ass decided to make the HSPI miso pin 12, which is a strapping pin.

Also tried all other 1.24.1 ports in the generic esp32-s3: 4mb flash and Support for Octal-SPIRAM. No errors were seen on any of the ports. This does usually fix WiFi issues. Am able to preform scan() properly. Probably similar to my issue #15602

### Code of Conduct

Yes, I agree

yfyungit, 2025-02-22:
You can check whether the router has WIFI6 turned on. If it has WIFI6, you can try to use the compatible mode WIFI5. I had difficulty connecting to WIFI6 with mircopython under esp8266 before, so I had to turn on WIFI5 compatibility.

PsuFan, 2025-02-22:
It does have WiFi 6 but I’ve never heard of such a setting. Like I said, I have lots of other esp devices that connect fine.

PsuFan, 2025-02-22:
Ugh GitHub auto close

PsuFan, 2025-03-02:
Tried another duplicate module that came in a three pack and did NOT have any issues connecting to WiFi... Assuming DOA module or from when I soldered header.