micropython/micropython #16779: RP2: Pico 2 W Not connecting with thread running

coencoensmeets, 2025-02-18:
### Port, board and/or hardware

RP2 - Pico 2 W

### MicroPython version

MicroPython v1.25.0-preview.229.g0d46e45a1 on 2025-01-27; Raspberry Pi Pico 2 W with RP2350

### Reproduction

```import network
import time
import _thread
import urequests

SSID = ""
password = ""

def connect_normal():
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.connect(SSID, password)
    print("Connecting to network")

    start_time = time.time()
    while not wlan.isconnected() and time.time() - start_time < 20:
        print("Waiting for connection...")
        time.sleep(2)

    if wlan.isconnected():
        print("Connected to network")
    else:
        print("Failed to connect to network")

    print("Sleep for 2 seconds")
    time.sleep(2)
    wlan.disconnect()
    
def main_thread():
    connect_normal()

def second_thread():
    for (i) in range(10):
        print("Second thread")
        time.sleep(2)

_thread.start_new_thread(second_thread, ())
main_thread()
```

Other things tested but not working:
- Extending the 20-second time limit.
- Putting `wlan = network.WLAN(network.STA_IF)`, `wlan.active(True)`, and `wlan.connect(SSID, password)` before starting a thread.
- Switching the functions `connection_normal` and `second_thread` to run in the other thread. So the connection is made on the newly opened thread.
- Building firmware with the latest Pico SDK develop
- Adding `rp2.country("NL")` and `network.country(country)`.

### Expected behaviour

Expected to connect to the network and print `Connected to network`.

### Observed behaviour

No connection is established after the set time of 20 seconds. (Time is irrelevant as it never seems to connect). When the line `_thread.start_new_thread(second_thread, ())` is commented out, a connection is found almost every single time.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Gadgetoid, 2025-02-19:
There are stability patches for the WiFi in Pico SDK 2.1.1 (though they are not really mentioned in detail and you need to delve into closed issues and commits) - https://github.com/raspberrypi/pico-sdk/releases/tag/2.1.1

If you can build MicroPython, I'd suggest bumping the SDK and re-trying your tests.

I do suspect this is a different issue, though, (something something IRQs?) threads have been pretty fraught on RP2 and I lose a day every time something like this comes up.

Gadgetoid, 2025-02-19:
I think I can confirm that something is going on here.

The second thread doesn't have to do anything, since merely starting it will fire up core1 and leave it in a wait loop.

I'm using a vanilla preview build and seeing output like those below. If I remove the thread... all good.

Something something interrupts?

```
MPY: soft reboot
Second thread
[CYW43] F2 not ready
Traceback (most recent call last):
  File "<stdin>", line 40, in <module>
  File "<stdin>", line 32, in main_thread
  File "<stdin>", line 14, in connect_normal
OSError: [Errno 1] EPERM
```

```
MPY: soft reboot
Second thread
[CYW43] do_ioctl(2, 268, 68): timeout
[CYW43] do_ioctl(2, 20, 4): timeout
[CYW43] do_ioctl(2, 22, 4): timeout
[CYW43] do_ioctl(2, 165, 4): timeout
[CYW43] do_ioctl(2, 26, 36): timeout
```

```
MPY: soft reboot
Second thread
[CYW43] do_ioctl(2, 263, 20): timeout
[CYW43] do_ioctl(
```

```
MPY: soft reboot
Second thread
[CYW43] core not in reset
[CYW43] core not in reset
[CYW43] core not in reset
[CYW43] HT not ready
```

```
MPY: soft reboot
Second thread
[CYW43] F2 not ready
```

coencoensmeets, 2025-02-19:
Working: MicroPython v1.23.0 on 2025-02-19; Raspberry Pi Pico W with RP2040
**Not** working: MicroPython v1.24.0 on 2025-02-19; Raspberry Pi Pico W with RP2040

So it's something between 1.23 and 1.24. As V1.23.0 does not have Pico2. This does not seem to be a Pico2 issue

coencoensmeets, 2025-02-19:
> I think I can confirm that something is going on here.
>
> The second thread doesn't have to do anything, since merely starting it will fire up core1 and leave it in a wait loop.
>
> I'm using a vanilla [...]

These error messages disappear for the Pico2 W with the newest version of Pico SDK. But then you just don't connect without error messages.

Gadgetoid, 2025-02-19:
> This does not seem to be a Pico2 issue

From some sleuthing I would concur. My working hypothesis (read: finger in the air guess) is that the GPIO interrupt for `CYW43_PIN_WL_HOST_WAKE` is being serviced by Core1 somehow, which is creating all sorts of chaos:

https://github.com/raspberrypi/micropython/blob/55b2720687b6bcd71b8a5c9a5d87bea53c57743a/ports/rp2/mpnetworkport.c#L53-L65

I'm not sure I'm right- though- since I can run Core1 *after* setting up wireless and requests seem to work fine. Though from experience it's very much the initial setup that's the most fragile part of the WiFi.

Gadgetoid, 2025-02-19:
Okay I was right- sort of. It's related to `PendSV_Handler` firing on either core and scheduled tasks not having an affinity.

The CYW43 driver is, for reasons I am not qualified to understand, extremely affinity sensitive so this hoses it completely.

Try this bruteforce fix:

```diff
diff --git a/ports/rp2/pendsv.c b/ports/rp2/pendsv.c
index 2c086f8942..ea79381360 100644
--- a/ports/rp2/pendsv.c
+++ b/ports/rp2/pendsv.c
@@ -133,6 +133,7 @@ void pendsv_schedule_dispatch(size_t slot, pendsv_dispatch_t f) {
 
 // PendSV interrupt handler to perform background processing.
 void PendSV_Handler(void) {
+    if(get_core_num() == 1) return;
 
     #if MICROPY_PY_THREAD
     if (!mp_thread_recursive_mutex_lock(&pendsv_mutex, 0)) {
```

DocMcC, 2025-02-19:
This may help, 
I have a program that connects in the main thread and then it opens a socket (http) webserver.  The listner is in a new thread.  It did work with a Pico 1 W then was making changes and updated the firmware to 1.25 and as soon as socket.accept() is called the wifi drops the connection/crashes.

with Pico 2 same thing - as soon as socket.accept() is called within a thread the wifi connection dies.

At one point i was getting the following error in Thonny: [CYW43] do_ioctl(2, 263, 75): timeout and [CYW43] do_ioctl(2, 263, 16): timeout - when the socket.accept() within the thread was called - I had some debugging code at the time, though removed it and now can't replicate the errors (note I don't get any errors now, just the device stops responding to ping as soon as the thread is called)

Code below:

import network
import _thread
import time
import socket

def webpage():
    global FeedHour, FeedMin, Timebet, NextFeedDate, LastFeedDate, FeedQuantity, FeedDay, TargetTemp, CurrentTemp
 
    #Template HTML
  
    html = f"""
        <!DOCTYPE html>
        <html>
        <body>
        <form action="./UpdateVal">
        Next Feed Date: {NextFeedDate}<br></br>
        Last Feed Date: {LastFeedDate}<br></br>
        Current Temp: {CurrentTemp}c<br></br>
        Day for Next Feed:<input type="text" id="FeedDay" name="FeedDay" value="{FeedDay}" /><br></br>
        Hour for feed:<input type="text" id="FeedHour" name="FeedHour" value="{FeedHour}" /><br></br>
        Minute for Feed:<input type="text" id="FeedMin" name="FeedMin" value="{FeedMin}" /><br></br>
        Hours between Feeds:<input type="text" id="Timebet" name="Timebet" value="{Timebet}" /><br></br>
        Feed Quantity:<input type="text" id="FeedQuantity" name="FeedQuantity" value="{FeedQuantity}" /><br></br>
        TargetTemp:<input type="text" id="TargetTemp" name="TargetTemp" value="{TargetTemp}" /><br></br>
        <input type="submit" value="Submit" />
        </form>
        
     
        </body>
        </html>"""
    return str(html) 
  
def setUpSocket():
    addr = socket.getaddrinfo('0.0.0.0', 80)[0][-1]
    s = socket.socket()
    s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    s.bind(addr)
    s.listen()
    return s

def serve(s,id):
    global FeedHour, FeedMin, Timebet, NextFeedDate, LastFeedDate, FeedQuantity, FeedDay, TargetTemp, CurrentTemp, Updated
    
    while True:

        client = s.accept()[0]
        request = client.recv(1024)
        request = str(request)
 
        try:
            request = request.split()[1]
            FeedDayNew= request.split("=")[1].split("&")[0]
            FeedHournew = request.split("=")[2].split("&")[0]
            FeedMinnew = request.split("=")[3].split("&")[0]
            Timebetnew= request.split("=")[4].split("&")[0]
            FeedQuantity=request.split("=")[5].split("&")[0]
            TargetTemp=request.split("=")[6]
 

            Updated=True;
        except IndexError:
            pass
     
        html = webpage()
        client.send(html)
        client.close()
    
def webserver():
    print("Webserver")
    s = setUpSocket() 
    _thread.start_new_thread(serve,(s,1))

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect("SSID", "PASSWORD")
count=0
while wlan.isconnected() == False:
    print('Waiting for connection...')
    count=count+1
    print(str(count))

    time.sleep(1)
    count=count+1
    if count==30:
       machine.reset()

ip = wlan.ifconfig()[0]
print(f'Connected on {ip}')
print(str(time.localtime()))
webserver()

while 1:
    tst=1

Hope this helps in finding the issue.

Gadgetoid, 2025-02-19:
> with Pico 2 same thing - as soon as socket.accept() is called within a thread the wifi connection dies.

AIUI CYW43/Wireless features are not thread safe and you should probably be using async.

IMHO Core1 availability on the Pico is a curiosity at best, and a vicious, useless footgun almost all of the time. It's better MicroPython threading is disabled and that core used for some performance critical baked-in C driver code.

Gadgetoid, 2025-02-19:
Why I do this to myself I don't know, but I did a little experimental affinity patch to pendsv and .... well, famous last words: It works for me! - https://github.com/micropython/micropython/pull/16788

DocMcC, 2025-02-20:
> Why I do this to myself I don't know, but I did a little experimental affinity patch to pendsv and .... well, famous last words: It works for me! - [#16788](https://github.com/micropython/micropython/pull/16788)

What do us mere mortals have to do test?  (sorry not that familiar with dev branches) or will your change make it into the next "release"   - thanks

Gadgetoid, 2025-02-20:
> What do us mere mortals have to do to test? (sorry not that familiar with dev branches) or will your change make it into the next "release" - thanks

That's a fair point!

(I don't know about making it into a release since I'm a semi-frequent contributor and bug finding pest, rather than a maintainer, but there will likely be "a fix" one way or another.)

These builds include the latest SDK and the pendsv affinity fix from my branch:

Pico 2 w:

[firmware-rp2350-pendsv-fix.uf2.zip](https://github.com/user-attachments/files/18883248/firmware-rp2350-pendsv-fix.uf2.zip)

Pico W:

[firmware-rp2040-pendsv-fix.uf2.zip](https://github.com/user-attachments/files/18883285/firmware-rp2040-pendsv-fix.uf2.zip)

I'm still seeing WiFi connection issues about 50% of the time, but that's with or without core1 enabled so I'll chalk that up to my usual problems with Pico W.

DocMcC, 2025-02-20:
Thanks so much, will take a look.  In the meantime I converted my program over to Asyncio and it is working fine (though not as responsive when I had the threads working....

Will give your firmware a bash and see if it works for me.

Much appreciated.

PR #16915 (2025-03-26): rp2: Fix lost CYW43 WiFi events when using both cores.
### Summary

Closes #16779. Thanks to @coencoensmeets for a clear and easy to reproduce bug report, and to @Gadgetoid for doing some deep digging to zero in on the root cause.

There's a very odd but predictable sequence of events that breaks Wi-Fi when using both cores:

1) CPU1 calls `pendsv_suspend()` - for example sleep() causes a softtimer node to be inserted, which calls `pendsv_suspend()`.
2) CYW43 sends wakeup IRQ. CPU0 GPIO IRQ handler schedules PendSV and disables the GPIO IRQ on CPU0, to re-enable after `cyw43_poll()` runs and c [... cut]