micropython/micropython #16792: RP2 2040 MQTT TLS

MyRaspberry, 2025-02-21:
### Port, board and/or hardware

RPI PICO W

### MicroPython version

after MP 1.23 upgrade to newer version
HIVEMQ TLS login fail
also last test
RPI_PICO_W-20250219-v1.25.0-preview.304.g1034b1755.uf2

fallback RPI_PICO_W-20240602-v1.23.0.uf2 works

use lib/
umqtt.simple.py 1.5.0

### Reproduction

import ssl
context = ssl.SSLContext(ssl.PROTOCOL_TLS_CLIENT)
context.verify_mode = ssl.CERT_NONE

mqtt_client = MQTTClient(
            client_id=b"somethingreallyrandomanduniquePICOW",
            server=secrets.RMQTTBROKER ,
            port=0,
            user=secrets.RMQTTUSER,
            password=secrets.RMQTTPW,
            keepalive=7200,
            ssl=context,
            )

mqtt_client.connect(clean_session=True, timeout=None) # fails with newer MP versions

### Expected behaviour

login to HIVEMQ TLS account

works with MP 1.23.0
but not with newer versions

### Observed behaviour

File "/lib/umqtt/simple.py", line 66, in connect
OSError: -2

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

jonnor, 2025-03-03:
Hi and thanks for the report. Does it work on MicroPython 1.24.1 ? Can you identify in which version it stopped working?

MyRaspberry, 2025-03-05:
sorry i not recall that specific version, 
i deleted the not working  1.24 UF2 s

felixdoerre, 2025-03-22:
So what is in line 66/65 of the umqtt that you are using? To me this looks like call to `getaddrinfo` that fails. So the connection already fails when the DNS lookup is performed, far before we do anything with TLS. Can you lookup any other host?

MyRaspberry, 2025-03-23:
@felixdoerre 
sorry HIVEMQ is my only TLS broker

just info
i converted project using Arduino IDE 2
and have
PICO W
PICO 2W
using PubSubClient.h and WiFiClientSecure
and 
ESP32s
ESP32S3
using internal <MQTT.h> and WiFiClientSecure

talking to that HIVEMQ remote broker
( that is what i call my 6$ telemetry )

but my normal site uses local RPI4 Mosquitto ( no TLS )
and Node-Red ( besides local database and operation ) 
to make a copy:
local Mosquitto to HIVEMQ
( that i call 66$ telemetry )
________________________
just last night i watched from Thailand my friend in USA
run a PICO 2W & INA260 on 15A
[via MQTT explorer](https://engineering-news.org/kllfusion01/downloads/PICO2W_INA260_MQTTexplorer.png)
_________________________
Circuit Python and Micro Python cost me too much time
( about 2 years )
about that MQTT TLS question 
and in both cases it looks like that the system ?timing?
is changed at an upgrade
and TLS not work anymore;
but i understand that you fight for every millisecond in the background loop

why not make a standard test
new mayor release with current umqtt and TLS
to a free remote broker you trust.

peterhinch, 2025-03-23:
Please see [this recommendation](https://github.com/orgs/micropython/discussions/16957#discussioncomment-12575635) re HiveMQ.

felixdoerre, 2025-03-23:
> sorry HIVEMQ is my only TLS broker

The problem you are facing seems to be related to basic network configuration. So I believe just opening any url (e.g. fetching "https://github.com" with requests), or a TCP connection to any host will face the same problem.

> is changed at an upgrade
> and TLS not work anymore;

I think, based on the error message, that it is highly unlikely to be TLS but already the DNS lookup even before the TLS connection. I've just did a build from master (for my pico-W) and tested an umqtt+TLS connection from there and it works flawlessly as before.

I don't want to encourage not using TLS, but if you believe TLS is the culprit, did you try connecting without TLS?

> why not make a standard test
> new mayor release with current umqtt and TLS

There is a test case to check a tls connection with a known-good host: https://github.com/micropython/micropython/blob/master/tests/net_inet/ssl_cert.py

Libraries from the micropython-lib repo don't have test cases in the main repo, but as I said, the problem you are facing happens before anything umqtt-specific.

MyRaspberry, 2025-03-23:
> Please see [this recommendation](https://github.com/orgs/micropython/discussions/16957#discussioncomment-12575635) re HiveMQ.

aha,
but wouldn't i see that error msg then?
and why it works with the old MP version?

dpgeorge, 2025-03-23:
> and why it works with the old MP version?

It could be due to IPv6 being enabled on the rp2 port since v1.24.0.  This could explain the networking error and inability to resolve the host name.

That said, it should still work like it did before.  I would try other basic TCP tests to try and narrow down the issue, like:
```py
>>> import requests
>>> r = requests.get('http://micropython.org/ks/test.html')
>>> r.text
```

MyRaspberry, 2025-03-24:
@dpgeorge 
______________________________________________________________________________
MPY: soft reboot
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040

Type "help()" for more information.

>>> import requests
>>> r = requests.get('http://micropython.org/ks/test.html')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "requests/__init__.py", line 182, in get
  File "requests/__init__.py", line 76, in request
OSError: -6
>>> r.text
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'r' isn't defined
>>> 
______________________________________________________________________________
MPY: soft reboot
MicroPython v1.25.0-preview.304.g1034b1755 on 2025-02-19; Raspberry Pi Pico W with RP2040

Type "help()" for more information.

>>> import requests
>>> r = requests.get('http://micropython.org/ks/test.html')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "requests/__init__.py", line 201, in get
  File "requests/__init__.py", line 79, in request
OSError: -6
>>> r.text
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'r' isn't defined
>>> 
______________________________________________________________________________

dpgeorge, 2025-03-24:
So your network connection is working (although you are setting a static IP... might be worth not doing that, just let it get DHCP).

You can also try a TLS test by using the same test URL but with https:
```py
import requests
r = requests.get('https://micropython.org/ks/test.html')
print(r.text)
```

MyRaspberry, 2025-03-24:
@dpgeorge 

ok i try:
**main.py**
```
import requests
# r = requests.get('http://micropython.org/ks/test.html')
r = requests.get('https://micropython.org/ks/test.html')
print(r.text)
```

and see:

```
MPY: soft reboot
try connect WIFI  kll-wlan_2.4G
('192.168.1.215', '255.255.255.0', '192.168.1.1', '8.8.8.8')
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
  File "requests/__init__.py", line 201, in get
  File "requests/__init__.py", line 79, in request
OSError: -2
>>> 
```

MyRaspberry, 2025-03-25:
@dpgeorge 

> static IP... might be worth not doing that, just let it get DHCP
we have multiple devices with web-page each
like

[image]

and use fixIP like ...216 with MQTT Topic _/D216/data

felixdoerre, 2025-03-25:
Hmm, maybe setting dns statically does not work. Can you try adding `network.ipconfig(dns="8.8.8.8")` after WiFi is connected, maybe after  `print("WiFi status: " + str(wlan.ifconfig()))`?

MyRaspberry, 2025-03-25:
@felixdoerre 

hm, that changed something???

```
MPY: soft reboot
try connect WIFI  kll-wlan_2.4G
WiFi status: ('192.168.1.215', '255.255.255.0', '192.168.1.1', '8.8.8.8')
@felixdoerre add ipconfig dns 
Current time: (2025, 3, 25, 16, 35, 14, 1, 84)
Connected to MQTT Broker: 01d9986933e64070a549e134b27cd218.s2.eu.hivemq.cloud
```

felixdoerre, 2025-03-25:
Yes, we improved network-handling code, and it seems I introduced a bug here:

https://github.com/micropython/micropython/blob/master/extmod/network_lwip.c#L97

As `netutils_parse_ipv4_addr` only gets a byte-pointer, it does not care about the type and does not set the address-type of `ip_addr_t dns;` to ipv4, leaving it uninitialized and making the lwip implementation fail dns queries.

Because we read out the dns server in the same manner, `ifconfig()` does not show, that the dns server is not set correctly.

Maybe, forcing the type to ipv4 by adding `IP_ADDR4(dns,0,0,0,0)` before or after `netutils_parse_ipv4_addr` would solve this.

Josverl, 2025-03-25:
Let leave the issue open as there is apparently something that needs to be fixed in the network stack.

Happy that you have a workaround , but you uncovered something that needs fixing. 
Knowing that is a good thing 👍

MyRaspberry, 2025-03-26:
@Josverl

thanks..
my code failed first when SSL was changed from boolen to context
**Matt Trentini** twitter @matt_trentini
helped me to do above TLS variant
what worked until end of 1.23

i think there is need for a MP MQTT TLS example
but must be under the **umqtt.simple** examples?

i work on that code
[MQTT_tutorial_TLS.txt](https://engineering-news.org/kllfusion01/downloads/MQTT_tutorial_TLS.txt)

how i get that umqtt.simple maintainer to use it? a new GIT ISSUE ?
or can you link him from here?

Josverl, 2025-03-29:
I'm not sure how to assist.
If you think another example is needed then you can create a new example or perhaps better extend the current .Rst document with a link to your example and send a PR.

I would recommend asking for a review of your sample, as part of that PR.