micropython/micropython #16799: Raspberry Pi Pico2 W - Wifi fails to start if CPU is overclocked

chrisib, 2025-02-23:
### Port, board and/or hardware

RPI_PICO_2_W

### MicroPython version

MicroPython v1.25.0-preview.307.g4364d9411 on 2025-02-23; Raspberry Pi Pico 2 W with RP2350

### Reproduction

Save the following code onto the Pico 2 W and run it with either `CPU_FREQ = 300_000_000` or `CPU_FREQ = 150_000_000` uncommented.

When using the overclocked frequency, the following error is printed and the access point is not available to connect to: `[CYW43] Failed to start CYW43`

When using the default frequency the access point can be connected to and no errors are printed

```
import machine
import network
"""
When overclocked we get
```
  [CYW43] Failed to start CYW43
```

When using default CPU frequency
the wifi starts fine
"""
CPU_FREQ = 300_000_000  # overclocked
#CPU_FREQ = 150_000_000  # default

machine.freq(CPU_FREQ)

wifi = network.WLAN(network.WLAN.IF_AP)
wifi.config(
    ssid="pico2w_ap",
    channel=10,
    security=network.WLAN.SEC_OPEN,
)
wifi.active(True)
```

### Expected behaviour

WiFi should work if the CPU is overclocked

### Observed behaviour

When the CPU is overclocked, the wireless card firmware does not appear to initialize correctly:

```
[CYW43] Failed to start CYW43
```

### Additional Information

My testing setup has 2 I2C devices connected: an OLED display GP0 & GP1 and a battery-backed realtime clock on GP2 & GP3.  The code above does not initialize these devices, so I don't _think_ their presence should have any impact; it looks like simply changing the CPU frequency is enough.

Doing a little trial-and-error changing the frequencies in the script above it looks like 270MHz is the upper limit of what the wifi can handle. Is this a known limitation of the firmware?

### Code of Conduct

Yes, I agree

Gadgetoid, 2025-02-24:
Have you tested against Pico SDK 2.1.1? It has stability fixes for WiFi which *might* also work here.

There's a branch with it bumped (and the small required fix to CMakeLists) here: https://github.com/dpgeorge/micropython/tree/lib-pico-sdk-211

It should probably go without saying; but nothing is guaranteed when you overclock. It's not a MicroPython limitation specifically but the hardware- and thus the SDK on which the Pico port relies- isn't officially supported at anything but the stock clock. MicroPython kinda leaves the guardrails off a bit here and lets you discover your own board's limits, but aiui does not attempt to make any guarantees about overclocking.

There is, iirc, also a lower limit at which WiFi breaks down. The CYW43 driver is... spicy.

chrisib, 2025-02-25:
I tried out the fork you linked, but I'm still seeing the same `Failed to start CYW43` error if I try to create an access point or connect to my home router while the CPU is overclocked.

Gadgetoid, 2025-02-25:
It’s probably just pushing the CYW43 bus clock out of range then.

It *might* be possible to fix this in MicroPython by proportionally raising the clock divider. IE: the opposite to what is discussed here: https://github.com/raspberrypi/pico-sdk/issues/1392

It’s not a function that is currently exposed, but as part of RM2 support maybe it could be added to the hypothetical CYW43 driver module mentioned here: https://github.com/micropython/micropython/pull/16057#issuecomment-2557072434

Though it would perhaps make sense for overclocking using “machine.freq()” to take care of this.

chrisib, 2025-02-26:
Good news: the new `1.25.0-preview.314.g8ce7a58be` appears to work if I overclock to `250_000_000`; it still fails to start at `300_000_000`, but it seems like an improvement.