micropython/micropython #16830: ESP32-P4

kdschlosser, 2025-02-27:
### Description

Any plans for supporting the ESP32-P4. I have a development board being delivered today and I didn't know what the plans are. I had seen that CircuitPython is working on it so I thought I would ask what the plans are for MicroPython as I don't want to run CircuitPython...

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

dpgeorge, 2025-02-28:
MicroPython supports IDF 5.4 which supports the ESP32-P4.  So it's a matter of configuring a board profile for the P4 then trying to compile it.

I had a quick go at doing that, and there are a few things that need adjusting in the code, such as touchpad, USB, ADC and PWM.  They could be disabled to start with just to get something working.

kdschlosser, 2025-02-28:
right now the master branch of MicroPython works with IDF 5.4?

dpgeorge, 2025-02-28:
Yes, see #16565.

kdschlosser, 2025-02-28:
there is more than is needed that the code changes in that PR. 

This needs to be added...

```
#ifndef USB_SERIAL_JTAG_PACKET_SZ_BYTES
#define USB_SERIAL_JTAG_PACKET_SZ_BYTES (64)
#endif
```

There is also some kind of an issue related to the WiFi that has not been addressed...

There is a problem with nimble that has not been addressed.

https://github.com/micropython/micropython/issues/15287

dpgeorge, 2025-02-28:
> This needs to be added...
> 
> ```
> #ifndef USB_SERIAL_JTAG_PACKET_SZ_BYTES
> #define USB_SERIAL_JTAG_PACKET_SZ_BYTES (64)
> #endif
> ```

This was done in 5564f3042ca850e363e6644b3df0b006785fcf2d

> There is also some kind of an issue related to the WiFi that has not been addressed something with an enumeration I believe

That was done by the linked PR, commit 865a4c8bf6868ce488347a76f5f3db8e0dabd11d

> There is a problem with nimble that has not been addressed

AFAIK that's not a problem.

I just now successfully compiled `ESP32_GENERIC` using IDF 5.4.  It built fine.  Deployed to a device and WiFi and BLE work.

kdschlosser, 2025-02-28:
I had thought you guys were going to stop supporting multiple IDF versions. There  was a lot of cleanup that was done when moving away from IDF 4.4 and it appears that it is headed back in the same direction as when IDF 4.4. was supported. It would be far less of a headache to update IDF versions if you didn't have to add code to alter behaviors based on what version of IDF is being used. It would be far easier from a maintenance aspect that is for sure.

dpgeorge, 2025-02-28:
I don't know why you are complaining.

You can read the README to see which IDF versions are supported.

We officially build with 5.2.2.  And in order to have a smooth update path to latter versions we bring in support for them when someone spends the time and energy to contribute patches.  Then eventually we will thoroughly test with a new version and update the build server and CI with that version.

Please try to be constructive with your contributions, and next time test before you make claims about things not working.  Otherwise it wastes eg my time retesting things, like I did above.  From a maintenance point of view I'd rather be doing other things.

kdschlosser, 2025-02-28:
I'm not complaining. I thought that was what you guys had decided to do.

closed 2025-02-28 by kdschlosser

kdschlosser, 2025-02-28:
and the reason why I am questioning if because issues I open seem to get dismissed as "no problem"

https://github.com/micropython/micropython/issues/16441

and then there is a merge that fixes the problem that didn't exist..

https://github.com/micropython/micropython/pull/16581#event-16101061366

andrewleech, 2025-02-28:
> and the reason why I am questioning if because issues I open seem to get dismissed as "no problem"

There was no dismissal here, simply counter-reports that your issue couldn't be reproduced locally or in CI. 
Generally If it cant be reproduced it can't be fixed. More information was needed to find a root cause.

You actually illustrate my point perfectly, the problem was identified and fixed in 16549 by the person who saw the problem, in identifying and fixing the cause it made it possible to share the root cause to be addessed differently in https://github.com/micropython/micropython/pull/16581

That's how it usually works most smoothly here, most often the people who find an issue are the best placed to fix it. 

Similarly, when I need an additional board supported, I put in the effort to add support myself.

favnec5, 2025-05-14:
hello,
would it be possible to detail a small procedure for compiling Micropython on the P4 ? I've tried with idf 5.4, but it doesn't seem so easy.

favnec5, 2025-05-15:
I thought we were at a stage afterward, since you had crossed it higher on your side.
i'm not out of the woods ... i will try too. Thanks

favnec5, 2025-05-15:
> > would it be possible to detail a small procedure for compiling Micropython for the P4
> 
> It's not that straightforward. You need to start with a board definition like `ESP32_GENERIC` and add this line to its `mpconfigboard.cmake`:
> 
> ```
> set(IDF_TARGET esp32p4)
> ```
> 
> and then try compiling and fix errors as they appear.

I thought we were at a stage afterward, since you had crossed it higher on your side.
i'm not out of the woods ... i will try too. Thanks

favnec5, 2025-05-19:
Ok, compiled 👍 

Project build complete. To flash, run:
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32p4 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 16MB --flash_freq 80m 0x2000 build-ESP32_GENERIC/bootloader/bootloader.bin 0x8000 build-ESP32_GENERIC/partition_table/partition-table.bin 0x10000 build-ESP32_GENERIC/micropython.bin
or from the "/mnt/c/Users/thomas_f/micropython/ports/esp32/build-ESP32_GENERIC" directory
 python -m esptool --chip esp32p4 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
bootloader  @0x002000    19200  (    5376 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  1652544  (  379072 remaining)
total                  1718080

But after flashing, M5Stack Tab5 reboot continually : " invalid header: 0x85a241e0 "
I don't know why :-/

favnec5, 2025-05-20:
It fixed :-)

esptool.exe --chip esp32p4 --port COM%com% --baud 46080 write_flash -z 0x002000 firmware.bin

[image]

mattytrentini, 2025-05-20:
> It fixed :-)

Are you able to share what you have? 😜

favnec5, 2025-05-20:
> > It fixed :-)
> 
> Are you able to share what you have? 😜

yes, of course :-)

This is not "perfect" at this time.
Pins / SoftI2C / RTC ...  OK
FLASH size : OK
memory is not is not fully recognized
Wifi works but i scan nothing (maybe not enough memory)
Bluetooth don't works.....

I'm trying to make more things work before push online

favnec5, 2025-05-24:
> > It fixed :-)
> 
> Are you able to share what you have? 😜

i did it ; https://github.com/micropython/micropython/pull/17354

On my M5 Tab5, it's more complicated because  it  is SDIO hosted ESP32C6 for wifi/ble.  I compile without error with the libraries and the corresponding idf parameters, but the wifi crash the P4. I'll open a discussion on it.

mattytrentini, 2025-05-26:
> But I don't know if all this will pass because I'm a bit clumsy with github.

Don't stress; some of us can help sort out that side of things! 😊 I have an M5Stack Tab5, an official ESP32 P4 dev kit (with display) and a Waveshare P4 Nano...I'll do some testing over the coming weeks. Thanks for your contributions!

jonnor, 2025-07-22:
There was a new MR here, https://github.com/micropython/micropython/pull/17356