micropython/micropython #16845: mpremote rmdir /   - Doesn't work

ZacharyACoon, 2025-03-03:
### Port, board and/or hardware

esp32c6 super micro

### MicroPython version

ESP32_GENERIC_S3-20241129-v1.24.1.bin

### Reproduction

`mpremote rmdir /`

```
rmdir :/
mpremote: Error with transport:
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
OSError: [Errno 1] EPERM
```

### Expected behaviour

Expected to erase all files from micropython's root.

### Observed behaviour

`OSError: [Errno 1] EPERM`

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-03-03:
mpremote's commands are based on unix commands of the same name.  Under unix, `rmdir` only removes *empty* directories.  So, it's not expected that it would remove the contents of that directory first.

Instead, on unix one would use `rm -rf /*`.  So if we wanted this feature to remove everything, we would need to implement `rm -r` (recursive remove).

ZacharyACoon, 2025-03-03:
> mpremote's commands are based on unix commands of the same name. Under unix, `rmdir` only removes _empty_ directories. So, it's not expected that it would remove the contents of that directory [...]

Is there an expected process to sync a directory or clear out the storage so that I can write the files I expect to be there?

ricksorensen, 2025-03-03:
Checkour https://github.com/orgs/micropython/discussions/9802 for a discussion of ways to clear out filesystem.  Also, `mpremote` uses hashes to check if destination file is same as source and will not copy  if they are the same.

peterhinch, 2025-03-03:
`rm -r` would be *very* useful.

dpgeorge, 2025-04-09:
Recursive remove has now been implemented in `mpremote`.  I suggest using `rm -rv` to show exactly what was removed.

closed 2025-04-09 by dpgeorge