micropython/micropython #16908: rp2.bootsel_button() not working on Raspberry Pi Pico 2 W (RP2350)

sozoraemon, 2025-03-11:
### Port, board and/or hardware

RP2 port, Raspberry Pi Pico 2 W (RP2350)

### MicroPython version

3.4.0; MicroPython v1.25.0-preview.49.g0625f07ad.dirty on 2024-11-21
(name='micropython', version=(1, 25, 0, 'preview'), _machine='Raspberry Pi Pico 2 W with RP2350', _mpy=7942)

- On Raspberry Pi Pico (RP2040), `rp2.bootsel_button()` works as expected.
- On Raspberry Pi Pico 2 W (RP2350), `rp2.bootsel_button()` **always returns `1`**, regardless of whether the BOOTSEL button is pressed or not.
- Tested on **MicroPython v1.25.0-preview.49.g0625f07ad.dirty (2024-11-21)**.
- This issue **does not occur on RP2040**, suggesting that the BOOTSEL button may be wired differently on RP2350.
- GPIO scan (`GPIO0-29`) did not detect any changes when pressing the button.
- `SIO` registers (`GPIO_HI_IN` and `GPIO_IN`) also did not show a clear state change.
- Further investigation is needed to determine how to read the BOOTSEL button on RP2350.

### Reproduction

### Steps to Reproduce
1. Copy and paste the following code into a new file:
   ```python
   import sys, time, rp2
   print(f"sys.implementation._machine={sys.implementation._machine}")
   for i in range(5):
       print(f"rp2.bootsel_button() = {rp2.bootsel_button()}")
       time.sleep(1)

2.Run the script on Raspberry Pi Pico 2 W (RP2350).
3.Press and release the BOOTSEL button during execution.

Expected behavior
・On Pico (RP2040): rp2.bootsel_button() returns 1 when pressed, 0 otherwise.
・On Pico 2 W (RP2350): Expected the same behavior.

Actual behavior
・Pico (RP2040):  Works correctly.
・Pico 2 W (RP2350): Always returns 1, even when the button is not pressed.

### Expected behaviour

### Expected Behaviour
- On **Raspberry Pi Pico (RP2040)**:
  - `rp2.bootsel_button()` should return `1` when the BOOTSEL button is pressed and `0` otherwise.
  - Example output when pressing and releasing the button:
    ```
    rp2.bootsel_button() = 0
    rp2.bootsel_button() = 1  # Button pressed
    rp2.bootsel_button() = 1
    rp2.bootsel_button() = 0  # Button released
    ```
    
- On **Raspberry Pi Pico 2 W (RP2350)**:
  - Expected **the same behavior** as RP2040.

- This function worked correctly in previous versions of MicroPython on **RP2040**.
- If `rp2.bootsel_button()` is not supported on RP2350, an alternative API should be provided.

### Observed behaviour

Observed Behaviour
On Raspberry Pi Pico 2 W (RP2350), rp2.bootsel_button() always returns 1, even when the BOOTSEL button is not pressed.

Actual output on Pico 2 W (RP2350):

sys.implementation._machine='Raspberry Pi Pico 2 W with RP2350'
rp2.bootsel_button() = 1
rp2.bootsel_button() = 1
rp2.bootsel_button() = 1
rp2.bootsel_button() = 1
rp2.bootsel_button() = 1

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Gadgetoid, 2025-03-11:
I'm going to hazard a guess this is E9-related. See Errata E9 in the datasheet - https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf - on page 1357/1358 for details.

There may be a workaround. I'll poke it if I get a free minute.

Gadgetoid, 2025-03-11:
I may be eating my words. I think it could just be a lack of special casing for the RP2350. The example from which the bootsel code was cribbed has since been updated:

https://github.com/raspberrypi/pico-examples/blob/84e8d489ca321a4be90ee49e36dc29e5c645da08/picoboard/button/button.c#L42-L46

sozoraemon, 2025-03-11:
Thank you very much for investigating this issue and for creating the PR! 
I really appreciate your time and effort in identifying the cause and implementing a fix.  
Looking forward to seeing it merged!

dpgeorge, 2025-03-13:
Fixed by e75ffc3bccb8b213704f0ddd2ce82731ab2069ba

closed 2025-03-13 by dpgeorge

sozoraemon, 2025-03-14:
Thank you very much for investigating and resolving this issue! I appreciate the effort and time spent identifying the cause and implementing the fix. I have confirmed that the problem has been successfully resolved in the updated firmware.

dpgeorge, 2025-03-14:
> Thank you very much for investigating and resolving this issue! I appreciate the effort and time spent identifying the cause and implementing the fix. I have confirmed that the problem has been successfully resolved in the updated firmware.

Excellent, thanks for confirming!