micropython/micropython #16912: When miso=None is set, the default miso pin changed state.

kaso-0, 2025-03-11:
### Port, board and/or hardware

Raspberry Pi Pico W

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Raspberry Pi Pico W with RP2040

### Reproduction

RS = Pin(16, Pin.OUT)

spi = SPI(0,
    baudrate=400000,
    polarity=0,
    phase=0,
    bits=8,
    firstbit=SPI.MSB,
    sck=Pin(18),
    mosi=Pin(19),
    miso=None) 

RS.value(0)
time.sleep_ms(1)
RS.value(1)

### Expected behaviour

Expected to set up SPI without MISO, set the RS pin value to 0, and after 1 ms, set the RS pin value to 1.

### Observed behaviour

The SPI was set up correctly, but the RS pin's value never changed.

### Additional Information

If I set up the pin after initializing SPI, it works as expected, the used pin 16 is default pin for miso. I think this bug is on other types of RPIs Pico.

### Code of Conduct

Yes, I agree

robert-hh, 2025-03-12:
The key feature here is to differentiate in the code between `no argument given` and `miso=None`. For that, both the ESP32 and SAMD port set a small negative number as default value, instead of "None" at the other arguments. Then the case `miso=None` can be detected  and treated accordingly. In case of the RP2 port this is a simple change.

Gadgetoid, 2025-03-12:
> In case of the RP2 port this is a simple change.

I think I've got a commit that does this gracefully.

I found `machine.SPI()` not resetting pins back to the default value quite confusing, so I ended up with:

```
>>> machine.SPI()
SPI(0, baudrate=1000000, polarity=0, phase=0, bits=8, sck=18, mosi=19, miso=16)
>>> machine.SPI(miso=None)
SPI(0, baudrate=1000000, polarity=0, phase=0, bits=8, sck=18, mosi=19, miso=None)
>>> machine.SPI()
SPI(0, baudrate=1000000, polarity=0, phase=0, bits=8, sck=18, mosi=19, miso=16)
```

Which is not the way setting up pins typically works... I might be solving a problem that I'm imagining here 😆  will raise the PR.

robert-hh, 2025-03-12:
> Even more egregious, the configured SPI pins stick through a soft reset which is a heck of a nasty bug!

I had a message exchange with Damien about the behavior of soft reset. His initial intention was, that soft reset should only reset the MicroPython state and leave all peripherals in the state they are. But that is not always possible and would require, that all objects representing a peripheral are static. If a peripheral object is created allocating memory from the heap or is using such memory for buffers like UARR does, then this is not possible any more. The MicroPython heap is reset on soft reset. My approach would be that Soft Reset would reset the peripherals as well and the device is in the state it was after hard reset and setting up clocks and memory devices. This might be a problem with e.g. WiFi or other communication links, which take a while to connect.
That said, the behavior during Soft Reset is not defined yet as a general rule and specifically for the  devices.

Gadgetoid, 2025-03-12:
> My approach would be that Soft Reset would reset the peripherals as well and the device is in the state it was after hard reset and setting up clocks and memory devices.

That would very much be my preference. Even as someone who goes to lengths to *not* use `machine.SPI` (for better or worse), I think it makes a lot of sense from a beginner perspective to have peripherals reset back into a known good state. For hardware resources like PIO, DMA, interrupt handlers? - where we do reset them - it's essential or things just break or get weird very quickly.

I've dived further into SPI and how its behaviour relates (or doesn't) to the documentation over in #16917 and because code speaks louder than words I'm putting together another commit to fix some of that. I have not yet tackled the soft reset issue though.