micropython/micropython #16929: API for Backup RAM access

kwagyeman, 2025-03-14:
### Description

It would be useful to create an API in the machine module for backup RAM access. This would allow for folks to read/write to any backup RAM on an MCU. For example, the STM32H7 has 4KB of backup RAM that holds is value between restarts. This is quite useful for applications that need to pass basic info between deepsleep events. Writing to the flash causes issues in the field as it can get corrupted.

Any thoughts on how the API should look? I think committing to and loading a dict from the backup RAM works. E.g. turn a dict into json, encode that, then write to backup ram. And then for loading try to return that dict.

### Implementation

I would like to sponsor development of this feature

### Code of Conduct

Yes, I agree

dlech, 2025-03-14:
Maybe something like:

* https://docs.micropython.org/en/latest/library/machine.html#memory-access
* https://docs.micropython.org/en/latest/library/uctypes.html#uctypes.bytearray_at

These already allow access to arbitrary memory addresses. You could convert json to string and store it using these.

Or use https://docs.micropython.org/en/latest/library/uctypes.html#uctypes.struct to avoid the need to convert the data to/from json.

kwagyeman, 2025-03-14:
Oh, yes, I've already written the code to do this for a customer using our system who needed backup RAM support. However, I'm thinking it would be good to have a proper API for this to follow.

Direct memory access to the array as a bytearray() works. But, it's somewhat low-level.

kwagyeman, 2025-04-11:
I see that this was added to the RTC object for the ESP32:

```
.. method:: RTC.memory([data])

   ``RTC.memory(data)`` will write *data* to the RTC memory, where *data* is any
   object which supports the buffer protocol (including `bytes`, `bytearray`,
   `memoryview` and `array.array`). ``RTC.memory()`` reads RTC memory and returns
   a `bytes` object.

   Data written to RTC user memory is persistent across restarts, including
   :ref:`soft_reset` and `machine.deepsleep()`.

   The maximum length of RTC user memory is 2048 bytes by default on esp32,
   and 492 bytes on esp8266.

   Availability: esp32, esp8266 ports.
```

This is great. I can add support for other MCUs off this API.

closed 2025-04-11 by kwagyeman