micropython/micropython #16930: Pico 2 W: I2C does not detect or communicate with certain addresses (0x30, 0x41, 0x48)

sozoraemon, 2025-03-14:
### Port, board and/or hardware

Raspberry Pi Pico 2 W

### MicroPython version

[v1.25.0-preview.383.gd5aeca2e8 (2025-03-13) .uf2](https://micropython.org/resources/firmware/RPI_PICO2_W-20250313-v1.25.0-preview.383.gd5aeca2e8.uf2)

### Reproduction

1.Connect Raspberry Pi Pico 2 W or Pico W.
2.Run the following Python script:
from machine import Pin, I2C
import utime

I2C_ADDR = 0x30  # Problematic address
i2c = I2C(1, scl=Pin(7), sda=Pin(6), freq=100000)

print("=== I2C Scan Start ===")
devices = i2c.scan()
print("Detected devices:", [hex(dev) for dev in devices])

if I2C_ADDR in devices:
    print(f"Device detected at {hex(I2C_ADDR)}")
else:
    print(f"Error: Device not found at {hex(I2C_ADDR)}")

try:
    print(f"\nWriting to {hex(I2C_ADDR)}...")
    i2c.writeto(I2C_ADDR, b'\x00')
    print("writeto() successful")
except OSError as e:
    print(f"writeto() error: {e}")

utime.sleep(1)

### Expected behaviour

On both Pico W and Pico 2 W, the I2C scan should detect all connected devices, including addresses 0x30, 0x41, and 0x48.
writeto() should work successfully for all detected devices.

### Observed behaviour

On Pico W, the I2C scan correctly detects devices at addresses 0x30, 0x41, and 0x48, and i2c.writeto() works without any issues.

On Pico 2 W, the I2C scan fails to detect these same addresses, returning an empty list []. Attempting to communicate with these addresses using i2c.writeto() results in the error:
OSError: [Errno 110] ETIMEDOUT

However, other I2C devices such as SSD1306 (0x3C) and BME280 (0x76) work normally on both boards, indicating that the issue is specific to certain I2C addresses rather than a general I2C failure.

This suggests a firmware-level bug in Pico 2 W’s I2C implementation, which prevents communication with specific addresses while functioning correctly with others.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-03-14:
Do you have pull-up resistors of the correct value (around 4.7k ohm) on the SCL and SDA lines?  That's always the first thing to check.

Also, you can try using software I2C:
```py
from machine import Pin, SoftI2C

i2c = SoftI2C(1, scl=Pin(7), sda=Pin(6))
i2c.scan()
```
That's pretty robust and should always work.  If it doesn't, the likely problem is with the hardware.

sozoraemon, 2025-03-14:
The Pico-CapTouch-ePaper-2.9 (Waveshare) was working fine on Pico W and Pico 2, but it no longer works on Pico 2 W using the same code.

I also tried SoftI2C as suggested, but on Pico 2 W, i2c.scan() returns []. In contrast, on Pico W and Pico 2, it correctly detects [48, 72].

I have not used pull-up resistors, but considering that the same setup worked on Pico W and Pico 2, this suggests that there may be differences in the I2C behavior on Pico 2 W (RP2350).

Do you have any thoughts on what could be causing this issue?
Also, if there are any additional debugging steps I can try, please let me know.

Thank you for your help!

dpgeorge, 2025-03-14:
If `SoftI2C` doesn't work, then it is very likely a hardware issue.

> I have not used pull-up resistors, but considering that the same setup worked on Pico W and Pico 2, this suggests that there may be differences in the I2C behavior on Pico 2 W (RP2350).

The Pico 2 and Pico 2 W both use RP2350, so it's unlikely the RP2350 that's the issue here.

Can you try a different Pico 2 W?

Do you have an scope that you can capture the signals on the I2C lines?

sozoraemon, 2025-03-14:
I have tested this on three different Pico 2 W boards, and all of them show the same issue.

Unfortunately, I do not have an oscilloscope or logic analyzer to capture the I2C signals.
Is there any other debugging method I can try to further investigate this?

Thank you for your support!

mendenm, 2025-03-17:
@sozoraemon:  you explicitly say you are not using pull-ups.  These are required.  Sometimes you get lucky, and the bus loading happens to be such that it works, but it is only luck.  Maybe the rp2350 has slightly different drive or bus capacitance on its board.  Try pull-ups.

sozoraemon, 2025-03-18:
Thank you for pointing out the pull-up resistors. I checked the schematic and found that 10kΩ pull-up resistors are connected to Touch_SDA (GP6) and Touch_SCL (GP7). I’m not very familiar with this, so I may not fully understand everything, but it seems that the issue is occurring even with these pull-ups in place. I apologize for any confusion, and I would appreciate any further advice you can provide.
https://files.waveshare.com/upload/9/9f/Pico-CapTouch-ePaper-2.9.pdf

robert-hh, 2025-03-18:
What is the level of GP16 (touch RST)? If that one is low, the touch device will not respond.

sozoraemon, 2025-03-18:
Thank you for your suggestion.
I checked the level of GP16 (touch RST), and it is High before and after the reset sequence.

I also tried executing the ICNT_Reset() function as suggested, but I2C scan still returns an empty list ([]) after the reset.

Using the same setup, the I2C scan results were as follows:
On Pico 2, the scan after reset detected devices at 0x48 and 0x41.
On Pico 2 W, the scan after reset returned an empty list.

robert-hh, 2025-03-18:
That is really strange, since Pico 2 and Pico 2W at the same firmware revision use the same I2C code. You could try a few things:
- measure the analog voltage level at SCL and SDA after the I2C object has been created. It should be close to 3.3V. I you do not have a multimeter, you can use the ADC input of another (or the same) Pico.
- Try usigng a very low I2C freq, like 1 kHz.
- Use a Pico as logic analyzer and sample the activity. The firmware from https://github.com/dotcypress/ula.git turns  a Pico into a logic analyzer frontend, controlled by the `pulseview` software.

robert-hh, 2025-03-19:
Just a question: Which connections do you have at J6 of the Waveshare display board?