micropython/micropython #16943: docs: The old forum is down

peterhinch, 2025-03-15:
### Description

https://forum.micropython.org/ produces a 404

### Code of Conduct

Yes, I agree

Josverl, 2025-03-15:
Does anyone know what the tech backend of the forum was/is? That would be a first step to assess the options to migrate the content somewhere else.

dpgeorge, 2025-03-15:
It works for me, if you do a reload...

It uses phpBB.

Josverl, 2025-03-16:
Ive set up a simple monitor to check availability from different locations,
and there is something flakey to say the least.

[image]

peterhinch, 2025-03-16:
Yes, it's 404 here (on two different browsers). It sometimes reappears briefly. See also https://github.com/orgs/micropython/discussions/16937

dpgeorge, 2025-04-07:
I had a look at the server logs, and it seems like a DDoS attack maybe?  Or just a lot of invalid/random traffic.

For `micropython.org` the stats for the requests and status codes are:
```
Time start:     2025-04-06 00:10:17
Time end:       2025-04-07 04:35:34
Total seconds:  102317.0
Total requests: 612715
200 : 573459 requests = 93.59%
206 : 828 requests = 0.13%
301 : 8315 requests = 1.35%
302 : 1680 requests = 0.27%
304 : 9790 requests = 1.59%
403 : 67 requests = 0.01%
404 : 13066 requests = 2.13%
416 : 101 requests = 0.01%
499 : 5257 requests = 0.85%
500 : 152 requests = 0.02%
```
That seems normal, about 6 requests per second, 94% of them return 200.

But for `forum.micropython.org` over the same time period it doesn't look good:
```
Time start:     2025-04-06 00:10:17
Time end:       2025-04-07 04:37:46
Total seconds:  102449.0
Total requests: 2623174
200 : 1011032 requests = 38.54%
206 : 35 requests = 0.001%
301 : 13656 requests = 0.52%
302 : 564 requests = 0.021%
304 : 66 requests = 0.002%
400 : 2 requests = 7.62e-05%
403 : 1840 requests = 0.07%
404 : 1100584 requests = 41.95%
499 : 495395 requests = 18.88%
```
That's a huge number of 404 and 499 responses.

Those 404/499 requests seem to be randomly generated phpBB URL that don't resolve to anything.

For now I've disabled the forum, although these random requests keep coming in (so it's definitely not real people accessing the site for those 404/499s).

Josverl, 2025-04-07:
I am responsible for 10 HTTP gets every 15 minutes for the Azure App Insights probes, 15m is the max interval I can set for forum and search  , and I check from different continents as there were significant differences in availability.
I suspect the forum may be under an investigative attack looking for vulnerabilities in handling these strange urls.

and you are correct that during the recent period there have been issues again

[image]

Josverl, 2025-04-07:
Possibly related : https://app.opencve.io/cve/?vendor=phpbb

CVE-2001-1471 | 1 Phpbb | 1 Phpbb | 2025-04-03 | 8.8 High

prefs.php in phpBB 1.4.0 and earlier allows remote authenticated users to execute arbitrary PHP code via an invalid language value, which prevents the variables 
 (1) $l_statsblock in prefs.php or 
 (2) $l_privnotify in auth.php from being properly initialized, which can be modified by the user and later used in an eval statement.

dpgeorge, 2025-04-07:
Thanks @Josverl for the extra info.  That plot does indicate that the attack started abruptly, with a good correlation of the date of that CVE being updated.

The forum uses a much more recent version of phpBB so shouldn't be impacted by that CVE.  Still, the traffic is annoying and slowing down some other micropython sites.

Josverl, 2025-04-07:
I can offer to run [Azure Front Door + WAF](https://learn.microsoft.com/en-us/azure/web-application-firewall/afds/afds-overview) Standard on my expense to help filter out that unwanted traffic if that would help.

dpgeorge, 2025-05-31:
@Josverl are you able to share recent plots for forum latency?

Also, are you constantly accessing http://micropython.org/ks/test.html ?  Someone is...

Josverl, 2025-05-31:
Currently traveling, I'll check when I get home.
IIRC The checks access a page every 15 minutes from 5 different locations globally. So about 20 requests/hour.

Josverl, 2025-05-31:
the esp8266 tutorial uses that URL, so that is possibly running on a lot of different boards.
docs/esp8266/tutorial/network_tcp.rst  :  ' response = http_get("http://micropython.org/ks/test.html")'

dpgeorge, 2025-05-31:
The access to `http://micropython.org/ks/test.html` is coming from a single IP address, around 5 times per second.

Josverl, 2025-06-01:
@dpgeorge 
the last 24 hours i've seen no issues

[image]

dpgeorge, 2025-06-02:
Thanks @Josverl for the data.  At least availability is not being impacted.

I can block the IP that's requisting `ks/test.html`, but would be interesting to know who it is.

Josverl, 2025-06-02:
always hard to tell with just an IP. Rate limiting would be best - but probably comes at a significant cost.
Ill keep the monitors running for now and notify you in case of alerts.

Josverl, 2025-06-27:
Did you receive the e-mail alert I set up, or did you find out yourself ?

[image]

Josverl, 2025-06-27:
I updated the config to send fewer alerts, and paused the alerting for now