micropython/micropython #16968: how can i  load a bytearray which contains a mpy file,and run it  in main.c ( ports/esp32)

XBCSmart, 2025-03-20:
### Description

I want to load a compiled .mpy file stored in a uint8_t array (named data with length _size) in main.c and run it. I've tried many methods but failed. I hope the maintainers can add a function load_from_mem_run(uint8_t data, size_t l_size) to pexec.h to enable a new way of running MicroPython. I am currently using version 1.25.0-preview." , thanks all

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

Josverl, 2025-03-20:
Some time ago I had a related Q on running a .mpy file via `mpremote run foo.mpy` that was essentially blocked on the same.

I can't find the thread, but IIRC then there was work underway to implement that.
Possibly @projectgus  knows

dpgeorge, 2025-03-20:
For this you can use `mp_raw_code_load_mem()`, which loads a .mpy file from an array and returns a function that can be executed.

There is also a helper function `ports/embed/port/embed_util.c:mp_embed_exec_mpy()` which does exactly what you want, you could copy that and use it in your C code.

XBCSmart, 2025-03-21:
> For this you can use `mp_raw_code_load_mem()`, which loads a .mpy file from an array and returns a function that can be executed.
> 
> There is also a helper function `ports/embed/port/embed_util.c:mp_embed_exec_mpy()` which does exactly what you want, you could copy that and use it in your C code.

thanks for your reply, i'll   take a stab at it

XBCSmart, 2025-03-21:
> Some time ago I had a related Q on running a .mpy file via `mpremote run foo.mpy` that was essentially blocked on the same.
> 
> I can't find the thread, but IIRC then there was work underway to implement that. Possibly [@projectgus](https://github.com/projectgus) knows

> Some time ago I had a related Q on running a .mpy file via `mpremote run foo.mpy` that was essentially blocked on the same.
> 
> I can't find the thread, but IIRC then there was work underway to implement that. Possibly [@projectgus](https://github.com/projectgus) knows

thanks your reply,  I will share it with you in a timely manner If I have any new progress

XBCSmart, 2025-03-21:
> Some time ago I had a related Q on running a .mpy file via `mpremote run foo.mpy` that was essentially blocked on the same.
> 
> I can't find the thread, but IIRC then there was work underway to implement that. Possibly [@projectgus](https://github.com/projectgus) knows
thanks your reply, I will share it with you in a timely manner If I have any new progress

jonnor, 2025-05-31:
@XBCSmart  Did the solution provided by dpgeorge above work?