micropython/micropython #16987: how to use the ESP-SR including: WakeNet、VADNet、MultiNet and other AI models of espressif.

zq19, 2025-03-21:
### Description

the espressif official provide few AI models, but only for c++ and arduino, how to use them with micropython?

### Code Size

the espressif official provide few AI models, but only for c++ and arduino, how to use them with micropython?

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

cnadler86, 2025-03-21:
Hi. I have binding for some esp AI Models at the moment. Here the [link](https://github.com/cnadler86/mp_esp_dl_models).
It is still experimental and not documented yet, but I think in the next days or weeks I will do this.

If you want to play with the take a look at the actions section, there you may find binaries as artifacts.

zq19, 2025-03-22:
amazing work! but I cant use it , i am new of coding. maybe I need to try the esp-idf. doc+AI may help me.

cnadler86, 2025-03-22:
Perhaps it would be good to move this to a discussion.

The I would suggest just to wait until I provide documentation on the binding.

cnadler86, 2025-03-22:
I took a quick look at the esp-sr, and it doesn't look too easy to bind (at least with my poor experience with Audio). So perhaps paitiance might be the key