micropython/micropython #16997: WebREPL command length limit

lukaskremla, 2025-03-23:
### Port, board and/or hardware

ESP32 S3, seems to affect all boards

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Generic ESP32S3 module with ESP32S3

### Reproduction

Try to execute [this script](https://github.com/lukaskremla/micropython-tools-jetbrains/blob/main/scripts/MicroPythonOptimized/scan_file_system_hashing.py) over WebREPL (Only copy it from the line containing imports, the license header is cut off before execution)

### Expected behaviour

The expected behavior is for the entire script to be accepted and executed just like over serial communication

### Observed behaviour

Only around 500 characters of the script should get written before a cut off

### Additional Information

I have looked for the WebREPL implementation in this repository. I only found the [modwebrepl.c](https://github.com/micropython/micropython/blob/master/extmod/modwebrepl.c) in the extmod folder. After quickly skimming through the file it seems that a 512 byte buffer is used for handling WebREPL input. The maximum command length that I can get executed over WebREPL actually roughly checks out with this buffer size.

Is this command length limitation intentional? Is there any way around it? 

The MicroPython Tools plugin for JetBrains IDEs I am working on relies on executing scripts similar to the one I shared above to provide its functionality. These scripts work just fine over serial communication, but they won't work over WebREPL due to this length limitation. 

The only plausible workaround I can think of is writing the scripts as files to the device and executing them from there. However, this would require a strong deviation from how commands are currently executed over serial communication. Also, due to the nature of embedded environments, I would like to avoid taking up space on the device's file system.

### Code of Conduct

Yes, I agree

Josverl, 2025-03-23:
not a solution - but possibly a workaround is to use minification of a readable & maintainable script - to a compressed and executable format.
 - `.mpy` can be one - but will tie to specific versions of micropython that work with that specific version of bytecode.
 - minification of the scripts using a tool such as might be a short-term workaround , for the above script that can bring it down from 843 Bytes to 487 Bytes ( depending on the options used), which may be just enough to unblock you. 
     - https://python-minifier.com/ & https://dflook.github.io/python-minifier/

lukaskremla, 2025-03-25:
> not a solution - but possibly a workaround is to use minification of a readable & maintainable script - to a compressed and executable format.
> 
> * `.mpy` can be one - but will tie to specific versions of micropython that work with that specific version of bytecode.
> * minification of the scripts using a tool such as `python-minifier`, might be a short-term workaround , for the above script that can bring it down from 843 Bytes to 487 Bytes ( depending on the options used), which may be just enough to unblock you.
>   
>   * https://python-minifier.com/ & https://dflook.github.io/python-minifier/

Thank you for the suggestions. I modified the build process and included a `python-minifier` step. I also made some more optimizations to the source scripts to try and shrink them down further. Unfortunately I still cannot get the entire hashing file system scan script executed over REPL.

`.mpy` could technically work, I plan on integrating `mpy-cross` into the plugin anyway. However, it adds more complexity to the script execution logic and is not a guaranteed fix since, beyond a certain file size, even compiling to bytecode won’t help.

The only way around this WebREPL limitation that I see is saving the scripts on-device, which is not ideal.

Josverl, 2025-03-25:
Did a quick PoC adding on to [webrepl_cli](https://github.com/Josverl/webrepl/blob/cli_windows/webrepl_cli.py), 
  

- [x] add windows terminal 
- [x] add send file to terminal , with delay/per line - dumb mode - no checks for throttling
- [ ] TODO: add [paste-mode / raw repl / raw-paste rep](https://docs.micropython.org/en/latest/esp8266/tutorial/repl.html#paste-mode)l p 

and it works 
- if you pace the inputs - so webrepl can keep up 
- if you are careful with spacing after indentation drop-offs ( this is here raw repl would come in ) 
- Adding paste mode and / or raw repl should also help in reducing the return traffic , and allow to reduce the delays needed

adding such a delay is always a bit of a hack - is it depends on the load on network and board (esp32 in this sample) 

<details><summary>Pics or it did not happen</summary>
<p>

https://github.com/user-attachments/assets/054130b4-61e8-4917-8185-1586f81ab400

```
MicroPython v1.24.1 on 2024-11-29; Generic ESP32 module with ESP32
>>> machine.freq()
160000000
``` 
</p>
</details>

lukaskremla, 2025-03-25:
I managed to get WebREPL working by utilizing careful chunking and delays. The below setup works for me on an ESP32 S3, I am not sure if the same delay will be enough for all boards, I will have to play around with this and see how far I can push things. Needless to say, this does ruin the file upload speed.

On boards that can handle it the built-in FTP uploads will come in handy. Those that cannot will hopefully only have to deal with smaller upload sizes.

I also found an older issue that touched on this - #2497. I was wrong about the cause of WebREPL's behavior, it seems that a ringbuf overflow is to blame.

<details>
<summary>Working code snippet</summary>

```kotlin
    private suspend fun sendCommand(command: String) {
        if (settings.state.usingUart) {
            mpyClient?.send(command)
        } else {
            command.lines().map { "$it\n" }.forEach { line ->
                // Convert to bytes for precise measuring
                val bytes = line.toByteArray(Charsets.UTF_8)
                var index = 0

                while (index < bytes.size) {
                    // Iterate over chunks of 255 bytes or whatever size remains
                    val endIndex = minOf(index + 255, bytes.size)

                    // Get chunk
                    val chunk = bytes.copyOfRange(index, endIndex)

                    // Convert back to string before sending
                    val chunkStr = String(chunk, Charsets.UTF_8)

                    // Send and wait
                    mpyClient?.send(chunkStr)
                    delay(200)

                    index = endIndex
                }
            }
        }

        mpyClient?.send("\u0004")
    }
```

</details>

Josverl, 2025-03-29:
> raw paste mode, is not actively delaying, but waiting until the board confirms that the free buffer is large enough

Interesting, is this documented, or is it your understanding of the webrepl server code?

felixdoerre, 2025-03-29:
This is documented for the repl as "raw paste mode" in general here: https://docs.micropython.org/en/latest/reference/repl.html#raw-mode-and-raw-paste-mode

Webrepl does not have any special code for the repl itself. It uses the same code that is used for the repl on serial/usb.

lukaskremla, 2025-04-16:
Thanks for the pointer. Raw paste mode did help. WebREPL is more stable with it and serial communication is also faster after implementing it.

Surprisingly, WebREPL is way slower than serial communication, but I suppose that raw paste mode pushes it as far as it can be pushed.