micropython/micropython #17016: slicing ranges gives different result vs python3

jepler, 2025-03-27:
### Port, board and/or hardware

unix

### MicroPython version

MicroPython v1.25.0-preview.418.g994751c251.dirty on 2025-03-26; linux [GCC 12.2.0] version

### Reproduction

1. Run a small snippet of code with standard python (3.11) & micropython
2. compare the results

### Expected behaviour

Python and micropython slice ranges in the same way

### Observed behaviour

```
$ ~/src/micropython/ports/unix/build-coverage/micropython -c 'print(range(-33, 18, 17)[-12:59:-2])'
range(-50, -50, -34)
$ python3 -c 'print(range(-33, 18, 17)[-12:59:-2])'
range(-50, 1, -34)
```

### Additional Information

I wrote a program to randomly slice random ranges. I ran it under python3 & micropython and selected one of the differences above.

All of the differences I found **were in zero-length slices** which mitigates the importance of the issue somewhat. However, these empty ranges still differ by their *end* values.

### Code of Conduct

Yes, I agree

jepler, 2025-03-27:
range slicing case generator (not exactly the same as the one I used to generate the original case):
```
def lcg():
    n = 19750408
    while True:
        n = (n * 1664525 + 1013904223) & 0xffffffff
        yield n % 120 - 60

rng = lcg().__next__
for _ in range(10000):
    r = range(rng() * 100, rng() * 100, rng() or 1)
    start = rng()
    end = rng()
    step = rng() or 1
    s = r[start: end: step]
    print(r, len(r), start, end, step, s, len(s))
```

andrewleech, 2025-03-27:
Interesting result, thanks for reporting with a clear reproduction!
Many types in micropython don't support slice with step !=1 eg. https://docs.micropython.org/en/latest/genrst/builtin_types.html#subscript-with-step-1-is-not-yet-implemented though for the reported limitations an exception is raised to inform the user.

jepler, 2025-03-27:
Here's a difference that doesn't involve step slicing:
```
$ python -c 'print(range(10)[1:0])'
range(1, 0)
$ src/micropython/ports/unix/build-coverage/micropython -c 'print(range(10)[1:0])'
range(1, 1)
```

jepler, 2025-04-26:
I think this was fixed in [8faa6ba](https://github.com/micropython/micropython/commit/8faa6bafdc76a6ee307551ef4d88fd2d54db04b2), closing.

closed 2025-04-26 by jepler