micropython/micropython #17018: SOC_TOUCH_VERSION_2 not define

Ansen88, 2025-03-27:
### Port, board and/or hardware

esp32 ports ESP32_GENERIC_S3 board

### MicroPython version

Micropython v1.24-release on 2025-3-27; 

### Reproduction

idf v5.4

cd ports/esp32
make BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT

### Expected behaviour

make completed

### Observed behaviour

FAILED: genhdr/qstr.i.last /home/ansen/work/src/micropython/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/genhdr/qstr.i.last 
/bin/sh esp-idf/main_esp32s3/CMakeFiles/qstr.i.last-93c470f.sh adef200ce74d197a
/home/ansen/work/src/micropython/micropython/ports/esp32/machine_touchpad.c:39:2: error: #error "Unknown touch hardware version"
   39 | #error "Unknown touch hardware version"
      |  ^~~~~

### Code of Conduct

Yes, I agree

cnadler86, 2025-03-28:
You need to use the actual Micropython master branch (or at least the colkit where Suppirt for IDF 5.4 was introduced) and not 1,24.x.

lhtien, 2025-04-03:
> You need to use the actual Micropython master branch (or at least the commit where Support for IDF 5.4 was introduced) and not 1.24.x.

Great!
After checking out the master branch, the build was successful. Thank you for your answer. However, I believe this bug should be fixed as soon as possible.

cnadler86, 2025-04-03:
This is not a bug. Micropython 1.24.1 does not support idf 5.4 (and I think 5.3 neither). Micropython 1.25 will. So, if you use the actual master branch you will use the 1.25 Preview.

closed 2025-05-06 by Ansen88