micropython/micropython #17020: Memory corrupted after micropython firmware programming

panther50500, 2025-03-27:
### Port, board and/or hardware

Arduino OPTA

### MicroPython version

I'm using Arduino Opta with MicroPython and two of them work fine. However, I ordered one online and encountered a problem after programming the firmware. Files in file explorers (Linux or Windows) are corrupted (special characters displayed and undefined files created). If I try to program it with Arduino, it seems to work fine. 
I returned it to the reseller who sent me another one but it has the same problem.
Someone else have this problem ?
Thanks
Frédéric

### Reproduction

try with :

ARDUINO_OPTA-20241129-v1.24.1.dfu (which works with another boards)
ARDUINO_OPTA-20250326-v1.25.0-preview.410.gdd7a950bb.dfu (which works with at least one other board)

### Observed behaviour

try with :

ARDUINO_OPTA-20241129-v1.24.1.dfu (which works with another boards)
ARDUINO_OPTA-20250326-v1.25.0-preview.410.gdd7a950bb.dfu (which works with at least one other board)

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2025-03-29:
That looks like the old problem of a filesystem on the board mounted at the PC and concurrent access by PC and board to the board's file system. The PC software assumes to be the only user and ignores changes made by the board. As long as you do not write to the file system from the boards and ALWAYS properly sync and eject the board from the PC it will work for a while.But more sooner than later you will end up with a corrupted board file system.  

The general rule is; do not use MSC.

P.S.: That's not limited to FAT. If you mount a LFS file system using e.g. littlefs-fuse, the situation & risk is the same.

panther50500, 2025-03-29:
thank you to take time to answered me,

The issue appear just after the micropython firmware download. If I read with the micropython interface or files explorer, without any writes on disk, the files are already corrupts

robert-hh, 2025-03-29:
You could try to re-format the board's file system using MicroPython:
1. Connect the board to the PC.
2. Eject the board's drive on the PC, if if is present. That is important!
3. Connect to the REPL using e.g. mpremote or another terminal emulator.
4. Format the drive as written in the docs at https://docs.micropython.org/en/latest/reference/filesystem.html#stm32-pyboard using:
```
import os, vfs, pyb
vfs.umount('/flash')
vfs.VfsFat.mkfs(pyb.Flash(start=0))
vfs.mount(pyb.Flash(start=0), '/flash')
os.chdir('/flash')
```
5. After the commands have finished hopefully without error, unplug the board and plug it into the PC again.

panther50500, 2025-03-29:
Thank you but I tried two times after reinitialise the arduino with factory default (memory_partionning.ino) but I have always the same issue.
- As you tell me I ejected the board's drive in the files explorateur
- With Arduino Lab for micropython I entered this lines without error :
  ``` import os, vfs, pyb
  vfs.umount('/flash')
  vfs.VfsFat.mkfs(pyb.Flash(start=0))
  vfs.mount(pyb.Flash(start=0), '/flash')
  os.chdir('/flash') ```
- I reboot by unplug/plug the usb port

robert-hh, 2025-03-29:
That's sad. You say that a different OPTA board is working? Is it the same model?
@iabdalkader Do you have any hint to proceed?

panther50500, 2025-03-29:
Yes, there is another Arduino Opta of the same model with Wi-Fi (AFX00002). With the other board, I had no problem. Everything works perfectly. 

But I don't think the problem is with the board; two boards with the same problem would be strange.would be weird.

iabdalkader, 2025-03-29:
I'm not sure. The Opta seems to ship with different QSPI chip, so maybe some variants have smaller sizes. I don't know of an easy way to read the QSPI ID, or capacity from Python without a debugger. I can ask around though on Monday.

panther50500, 2025-03-29:
In the documentation :
1MB RAM (programming)
2MB internal + 16MB Flash QSPIOk, thanks to your return

panther50500, 2025-03-31:
These are screenshots of the issue :

Freeze
[image]

or corrupted files :

[image]

facchinm, 2025-03-31:
Hi @panther50500 ,
we are analyzing the situation internally, will let you know as soon as we have news on what's causing the problem and hopefully a workaround