micropython/micropython #17035: Panic when calling timer.value() on an uninitialised timer (ESP32)

robert-hh, 2025-03-29:
### Discussed in https://github.com/orgs/micropython/discussions/17033

<div type='discussions-op-text'>

<sup>Originally posted by **SergeiFranco** March 29, 2025</sup>
I have encountered a peculiar problem:
if I call value() on a uninitialised timer I get a panic:
Here are steps to reproduce:
```
t0 = Timer(0)
t0.value()
```

Results in:
```
A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : v1.24.1 on 2024-11-29
IDF version : v5.2.2
Machine     : Generic ESP32 module with ESP32

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d6790  PS      : 0x00060330  A0      : 0x800dd400  A1      : 0x3ffcf370  
A2      : 0x3ffd3890  A3      : 0x00000000  A4      : 0x00020002  A5      : 0x00000000  
A6      : 0x0000005c  A7      : 0x3ffbc984  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000003  A11     : 0x00060323  A12     : 0x00000001  A13     : 0x0000000d  
A14     : 0x3ffd6f40  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x400d678d:0x3ffcf370 0x400dd3fd:0x3ffcf390 0x400e4935:0x3ffcf3b0 0x400e49fd:0x3ffcf3d0 0x40084d22:0x3ffcf3f0 0x400dd51f:0x3ffcf490 0x400e4935:0x3ffcf4e0 0x400e494a:0x3ffcf500 0x400f2763:0x3ffcf520 0x400f2a58:0x3ffcf5b0 0x400d6274:0x3ffcf5f0

ELF file SHA256: 4ac361b7d

Rebooting...
```

Note: works on any timer (not just 0).

The micro details are:
```
MPY version : v1.24.1 on 2024-11-29
IDF version : v5.2.2
Machine     : Generic ESP32 module with ESP32
```</div>

jonnor, 2025-04-25:
Thanks for reporting. And for the MR :)

dpgeorge, 2025-05-14:
Fixed by 1d4bf8ac40236e22d0a67b103b4330d1ee6659d4

closed 2025-05-14 by dpgeorge