micropython/micropython #17046: Enable MICROPY_EMIT_RV32 for ESP32-C6 but compile error "invalid arch" with decorator @micropython.native

bhcuong2008, 2025-03-30:
### Port, board and/or hardware

ESP32 C6

### MicroPython version

MicroPython 1.24.1

### Reproduction

1. Create a test file test.py inside frozen folder modules/test.py
2. Write a function with decorator @micropython.native
4. Build for target ESP32 C6

### Expected behaviour

Compile successfully.

### Observed behaviour

Compile failed with error "invalid arch"

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

agatti, 2025-03-30:
You may have an old build of `mpy-cross` lingering around, one that was built either before the RV32 native code was introduced, or that has RV32 native code support explicitly disabled in `mpy-cross/mpconfigport.h`.

Can you please delete `mpy-cross/build`, make sure that `MICROPY_EMIT_RV32` is enabled in `mpy-cross/mpconfigport.h`, rebuild `mpy-cross` and try again?

bhcuong2008, 2025-03-31:
I delete directory mpy-cross, then checkout again. I checked mpconfigport.h and it has define `#define MICROPY_EMIT_RV32           (1)` But the error still exists.

bhcuong2008, 2025-03-31:
Yes, with command, `./mpy-cross -march=rv32imc ../../ports/esp32/modules/_cinit.py`, it's ok. I will check again my environment setup. Thank you for your guide.

bhcuong2008, 2025-03-31:
My build is successful now. Thank you again. It seems that in the port ESP32, it only includes xtensa, `set(MICROPY_CROSS_FLAGS -march=xtensawin)`.

closed 2025-03-31 by bhcuong2008

agatti, 2025-03-31:
Glad to hear you got it working - that specific issue already has a PR for it: #16993.