micropython/micropython #17047: RP2350: PIO scripts fail, despite running on RP2040

peterhinch, 2025-03-30:
### Port, board and/or hardware

Pico 2

### MicroPython version

MicroPython v1.25.0-preview.408.gf315a376b on 2025-03-26; Raspberry Pi Pico2 with RP2350

### Reproduction

It has proved hard to produce a minimum repro, what follows is my best shot. The repro is quite fragile - minimal changes cause its behaviour to vary between working and failing.

Link GPIO numbers 1-18 and 2-17 and run the following:
```py
import rp2
from machine import Pin, SPI

@rp2.asm_pio(autopush=True, autopull=True)
def spi_count():
    out(x, 32)
    wait(0, pins, 2)  # CS/ True
    label("loop")
    jmp(pin, "good")
    jmp(x_dec, "loop")
    label("good")
    in_(x, 32)
    label("finish")
    jmp("finish")

class Test:

    def __init__(self, start_pin=0, sm_num=0):
        self._sm1 = rp2.StateMachine(
            sm_num,
            spi_count,
            in_base=start_pin,
            jmp_pin=start_pin + 1,
        )
        self._sm1.active(1)

    def go(self, cs, ck):
        sm = self._sm1
        sm.put(10_000_000)
        cs(0)  # Start
        ck(1)  # Stop
        
        assert sm.rx_fifo(), "Rx fifo is empty."
        print(sm.get())

cs = Pin(17, Pin.OUT, value=1)
ck = Pin(18, Pin.OUT, value=0)

t = Test()
t.go(cs, ck)
```

### Expected behaviour

Result on RP2040 is to print a number around 9999759.

### Observed behaviour

RP2350 produces an assertion fail. I assume that the RP2350 is failing to respond to the jump pin.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Gadgetoid, 2025-03-31:
I can confirm the RP2350 case, though I haven't tested on a 2040 to verify it works out of the box.

It seems to be happy if I add:

```
_ = Pin(1, Pin.IN)
```

I can golf it down to:

```python
import rp2
from machine import Pin, SPI

@rp2.asm_pio(autopush=True, autopull=True)
def spi_count():
    label("loop")
    jmp(pin, "good")
    jmp("loop")
    label("good")
    in_(pins, 32)

#_ = Pin(1, Pin.IN)  # Uncomment for RP2350

cs = Pin(17, Pin.OUT, value=1)
ck = Pin(18, Pin.OUT, value=0)

sm = rp2.StateMachine(
    0,
    spi_count,
    jmp_pin=1,
)
sm.active(1)
sm.put(0)
cs(0)  # Start
ck(1)  # Stop
assert sm.rx_fifo(), "Rx fifo is empty."
v = sm.get()
print(f"{v:032b} {v}")
```

Which fails consistently for me on a Pico 2 (W but running non-W firmware) and succeeds consistently on Pico (running `MicroPython v1.25.0-preview.430.g5eee5a67dc.dirty on 2025-03-28`)

I don't know if I'm testing the same thing as you here, but it would appear the initial state of input pins differs enough between RP240 and RP2350 to cause a problem here. It doesn't feel like `_ = Pin(1, Pin.IN)` should be needed at all!

Gadgetoid, 2025-04-01:
I had a suspicion it wouldn't be so simple, probably worth a documentation or code fix for the jump pin either way.

You're not running into the E9 errata on the RP2350 somehow, perhaps?

I had a pretty straight forward wiring set up on mine, but the only time it failed unexpectedly was when I forgot to move the wires between the Pico and Pico 2.

I can confirm that `_ = Pin(1, Pin.IN)` fixes your original test case for me, and that it otherwise fails very consistently.

Setting up pins and checking the wiring inexplicably causing it to run seems to be recreating `_ = Pin(1, Pin.IN)`. It doesn't matter if the jump pin is an input or an output, it appears, just that it's set up *somehow* to clear the pad isolation latch.

Is there anything else particular about your setup/wiring that might play a role?

peterhinch, 2025-04-02:
In both cases (RP2040, RP2350) I have a Pico plugged into a breadboard with three jumpers linking the board's SPI output to the input pins of my slave. I have tried reducing the clock speed, using SoftSPI, and checking the waveforms on a scope. I've also tried the `_ = Pin(1, Pin.IN)` fix. The one thing that is consistent is that the slave always works on RP2040 and never works on RP2350. Symptoms are consistent with failure to respond to the jump pin.

I have spent an unreasonable amount of time trying to achieve a repro - evidently my script has revealed one problem, but not the entire picture. I'm rather running out of ideas and am unsure how (or if) to proceed.

robert-hh, 2025-04-02:
Does it fail as well if you run the SPI master on a different board?

peterhinch, 2025-04-02:
@Gadgetoid Eureka!
```py
mem32[0x40038000 | 0x04] |= (1 << 6)
mem32[0x40038000 | 0x04] &= ~(1 << 8)
mem32[0x40038000 | 0x08] |= (1 << 6)
mem32[0x40038000 | 0x08] &= ~(1 << 8)
```
fixes my application. GPIO0 is the data pin - without applying the fix to that pin too, the application responded to the jump pin but DMA input produced no data.

Now working perfectly!

Gadgetoid, 2025-04-02:
D’oh! Definitely an input enable / isolation latch problem then, what a pitfall 🫠

Glad you’re sorted though!

Not sure what, if anything, we do about jmp_pin configuration.

peterhinch, 2025-04-03:
Just to clarify, the _only_ change I applied was the four pokes listed above. I did not apply `_ = Pin(1, Pin.IN)`. The pokes fixed the jump pin and the one input pin.

It seems to me that a PR is required to make the PIO usable for tasks requiring pin input, but you're the expert on the hardware :)

Gadgetoid, 2025-04-03:
> but you're the expert on the hardware :)

Haaaaa 🤣 more like determined full time amateur.

Josverl, 2025-07-22:
For other uses MicroPython mostly accepts either a pin, an int, (or a string.)

So I am in favor of fixing it.

Gadgetoid, 2025-07-23:
> Does this (and also `Pin(0, Pin.IN)`) fix the issue by itself?

From my tests yes, since it clears the isolation latch.

There have been some Pico SDK documentation tweaks relating to this gotcha which I requested here - https://github.com/raspberrypi/pico-sdk/issues/2387

IIRC I have a vague recollection that it's possible to tell if it pin has been configured and, if not, configure it. A sensible change for PIO in both the RP2040 and RP2350 case?

peterhinch, 2025-07-23:
> That's also surprising. So you say that:
StateMachine(..., jmp_pin=1)
doesn't work, yet:
StateMachine(..., jmp_pin=Pin(1))
does work?

The first does not work. I haven't tested the second. What does work consistently is
```py
StateMachine(..., jmp_pin=Pin(1, Pin.IN))
```
All input pins passed to the constructor must be initialised as `Pin(n, Pin.IN)`.

dpgeorge, 2025-07-29:
I can confirm the issue, using the test at the top of this issue.

The fix (for me) is to put `Pin(1, Pin.IN)` somewhere near the start, to make it a proper input pin.  I guess you also need to do `Pin(0, Pin.IN)` to make the data input pin work as well.

The reason for the issue (as discussed above) is that on RP2040 pins default to input (or at least readable by PIO), but on RP2350 they default to isolated.

I think the only thing we can do here is, for pins used as jmp pins, if they are isolated then turn off the isolation (and maybe also enable them as input).  Otherwise leave them alone (because the user probably configured them already).

For input data pins to PIO (used with `in_(pins, ...)`) we can't really do anything, because we don't know how many pins are used as the input.  It's up to the user to make sure they configure the pin(s) used with `in_` to the correct state.

Really the issue here feels like a documentation issue, but because the same code works fine on RP2040, I think it's fair enough to configure PIO jmp pins as input.  And also add something to the docs.