micropython/micropython #17048: ESP32S3 vendor specific requests

emard, 2025-03-30:
### Description

Is there possible to have some USB callback that would handle USB vendor specific requests?
I tried to develop but I'm stuck here, USB_REQ_TYPE_CLASS works but USB_REQ_TYPE_VENDOR
doesn't. If I can recompile micropython with this enabled, tell me how should I somehow
enable CFG_TUD_VENDOR and in what file for ESP32S3 SPIRAM OCT should I edit to enable this

currently practical problem is described here for example

  if stage == 1:  # SETUP
    # BUG USB_REQ_TYPE_VENDOR requests don't work
    if bmRequestType == USB_DIR_OUT | USB_REQ_TYPE_CLASS | USB_REQ_RECIP_DEVICE:
    #if bmRequestType == USB_DIR_OUT | USB_REQ_TYPE_VENDOR | USB_REQ_RECIP_DEVICE:
      # Data coming from host, prepare to receive it.
      return memoryview(usb_buf)[:wLength]

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

andrewleech, 2025-03-31:
Hi, do you mean this as a feature you'd like added to the `machine.USBDevice` API?

emard, 2025-04-01:
Yes we can say this is a feature and would like to be added.
First I thought it was a bug in my code so I started to write
testbenches and cheks and after 2 weeks found this is not
a bug but a missing feature :)

andrewleech, 2025-04-03:
I haven't used USB_REQ_TYPE_VENDOR myself in the past so not sure, but yes you could try adding 
``` c
#define CFG_TUD_VENDOR            (1)
``` 
to https://github.com/micropython/micropython/blob/master/shared/tinyusb/tusb_config.h

andrewleech, 2025-04-03:
Oh, that being said that config for tinyusb looks like it's for the internal handling of the class which shouldn't be needed at all for use with `machine.USBDevice`. So that's quite possibly not what you want, I'm not sure why it wouldn't work already with the dynamic / runtime usb interface. @projectgus do you know about handling of the vendor class type?

Just to be clear @emard are you trying to implement something like one of these examples https://github.com/micropython/micropython/tree/5eee5a67dc69f6b701e9a79b6fd1e047dfbd55a9/examples/usb but make it appear as a "vendor type device"? Or is the `USB_REQ_TYPE_VENDOR` message just to receive custom / vendor control messages?

emard, 2025-04-03:
What I really want to (try to) do is to make micropython
version of this
https://github.com/sipeed/RV-Debugger-BL702
and this is some minimalsitic implementation of ft2232
just enough for linux driver to recognize it as ft2232
and load default driver.
For this I need VENDOR requests supported
And looking at descriptor it is also 255 device
Id so it belongs to vendor specific class of devices I guess

projectgus, 2025-04-15:
> Or is the `USB_REQ_TYPE_VENDOR` message just to receive custom / vendor control messages?

TinyUSB routes vendor requests to a different C callback, `tud_vendor_control_xfer_cb()`. This isn't currently implemented in MicroPython's dynamic USB support, but I think implementing it should be relatively straightforward (I think on our side we could possibly route it to the existing control transfer callback).

emard, 2025-04-15:
This support would be great!
Having one callback is intuitive,

2 separate callbacks may be 0.1% faster
but 99.9% counter intuitive ;)

emard, 2025-04-19:
I can try it on esp32s3
is vendor request early support available
in latest micropython