micropython/micropython #17051: ESP32 UART receive interrupt cannot be triggered after adding timer.

Tinyu-Zhao, 2025-03-31:
### Port, board and/or hardware

esp32

### MicroPython version

MicroPython v1.24.0-dirty on 2025-03-13; M5STACK AtomS3 with ESP32-S3-FN8

### Reproduction

```python
import sys
import os, sys, io
from machine import UART
from micropython import schedule
from hardware import Timer
import micropython

if sys.platform != "esp32":
    from typing import Literal

class ASRUnit:
    DEBUG = True

    myprint = print if DEBUG else lambda *_, **__: None

    _COMMAND_LIST = {
        0x32: ["hello", None],
    }

    def __init__(self, id: Literal[0, 1, 2] = 1, port: list | tuple = None):
        self.uart = UART(id, tx=port[1], rx=port[0])
        self.uart.init(115200, bits=8, parity=None, stop=1)
        self.uart.irq(
            handler=self._handler,
            trigger=UART.IRQ_RXIDLE,
        )
        self.raw_message = ""
        self.command_num = 0
        self.is_recieved = False

    def _handler(self, uart) -> None:
        data = uart.readline()
        if data is not None and len(data) >= 5:
            self.myprint(("Received data: ", data))

micropython.alloc_emergency_exception_buf(100)

timer0 = None
i2c1 = None
motion = None
asr_0 = None

def asr_0_hello_event(args):
    global timer0, i2c1, motion, asr_0
    print("hello M5")

def timer0_cb_safe(arg):
    print("Timer event executed in main thread")

def timer0_cb(t):
    global timer0, i2c1, motion, asr_0
    schedule(timer0_cb_safe, None)

def setup():
    global timer0, i2c1, motion, asr_0
    asr_0 = ASRUnit(2, port=(7, 8))
    # timer0 = Timer(0)
    # timer0.init(mode=Timer.PERIODIC, period=2000, callback=timer0_cb)

def loop():
    global timer0, i2c1, motion, asr_0

if __name__ == "__main__":
    try:
        setup()
        while True:
            loop()
    except (Exception, KeyboardInterrupt) as e:
        try:
            from utility import print_error_msg

            print_error_msg(e)
        except ImportError:
            print("please update to latest firmware")
```

Run this Code

### Expected behaviour

Uncomment the timer0 related lines in the above code, and the UART interrupt will never be triggered.

```python
def setup():
    global timer0, i2c1, motion, asr_0
    asr_0 = ASRUnit(2, port=(7, 8))
    timer0 = Timer(0)
    timer0.init(mode=Timer.PERIODIC, period=2000, callback=timer0_cb)
```

### Observed behaviour

bad code 
```shell
Timer event executed in main thread
Timer event executed in main thread
Timer event executed in main thread
```

good code 
```shell
('Received data: ', b'\xaaU2U\xaa')
```

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2025-03-31:
Timer(0) is needed for UART.IRQ_RXIDLE. If you reconfigure it, the IRQ_RXIDLE interrupt will not work.
Solution: use a different timer for your code.

Tinyu-Zhao, 2025-03-31:
Thank you for your help, it solved my confusion. 

By the way, I think it might be better to write clearly in UART.iqr that it is using Timer(0). 

Thank you very much.

closed 2025-03-31 by Tinyu-Zhao

robert-hh, 2025-03-31:
I prepared already a sentence to the documentation.

Tinyu-Zhao, 2025-03-31:
> I prepared already a sentence to the documentation.

🌹