micropython/micropython #17052: mpremote fs cp  fails  and I don't understand why.

ZacharyACoon, 2025-03-31:
### Port, board and/or hardware

mpremote 1.24.1

### MicroPython version

mpremote 1.24.1

### Reproduction

copy a directory of files to root :/

### Observed behaviour

(emdr-v1-firmware) PS C:\Users\Zac\PycharmProjects\emdr-v1-firmware> mpremote fs cp -r .\root\. :/    
cp .\root\. :/
Up to date: ././app.py
Up to date: ././config.json
Up to date: ././emdr.py
Up to date: ././hardware_config.py
 ... copying  59% [###########---------]mpremote: Error with transport:
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-03-31:
Please state the hardware/board/port that you are running MicroPython on, and also your operating system, eg WSL.

The issue could be due to Windows serial port driver bugs.

jabelone, 2025-04-23:
I had a very similar issue happening to me. I could flash firmware fine but file uploads rarely worked and I kept getting all sorts of random errors, but mostly a syntax error similar to that one. Turns out the USB serial adapter I was using was bad (or maybe its driver was buggy). I bought a new one and it started working fine again after that.

Josverl, 2025-04-24:
I think the path handling gets confused by non-posix path delimiters (`\` ) as evident the incorrect paths starting with : `././`

can you try `mpremote fs cp -r ./root. :/` or `mpremote fs cp -r root. :/` ?