micropython/micropython #17072: TCP socket RESET's the connection when not reading the entire read buffer

stupid-stove, 2025-04-02:
### Port, board and/or hardware

W5500-EVB-Pico, Raspberry Pi Pico W

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; W5500-EVB-Pico with RP2040

### Reproduction

When not reading the entire TCP message, micropython will RESET the connection, instead of doing a proper TCP connection teardown.

Example:
```python
import network
from machine import SPI, Pin
import time
import asyncio

def w5x00_init():
    # https://github.com/Wiznet/RP2040-HAT-MicroPython/blob/main/Ethernet%20Example%20Getting%20Started%20%5BMicropython%5D.md
    spi=SPI(0,2_000_000, mosi=Pin(19),miso=Pin(16),sck=Pin(18))
    nic = network.WIZNET5K(spi,Pin(17),Pin(20))
    nic.active(True)
    nic.ifconfig('dhcp')
    while not nic.isconnected():
        time.sleep(1)
        print(nic.regs())
    print(nic.ifconfig())

async def handle_client_read_1000(r, w):
    await r.read(1000)
    w.write(b'HTTP/1.0 200 OK\r\nContent-Length: 2\r\nContent-Type: text/plain\r\n\r\nhi')
    w.close()
    await w.wait_closed()

async def server_async(handle_client):
    await asyncio.start_server(handle_client, '0.0.0.0', 80)
    while True:
        await asyncio.sleep(1)
        
w5x00_init()
asyncio.run(server_async(handle_client_read_1000))
```

### Expected behaviour

Reading parts of the TCP stream/reading the entire stream/not reading the stream should not change the behavior of writing into the stream.

When running the same code (except the network setup) in CPython it does a proper TCP connection teardown and does not send RESET.

### Observed behaviour

When doing
```sh
curl 192.168.2.127
```

we get a TCP connection teardown after the payload has been send (FIN-ACK):

[image]

But when sending a message longer then the buffer (here 1000B) via
```sh
curl 192.168.2.127/hiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiii

```
we get

[image]

I.e. it sends RESET instead of doing a proper connection teardown.

This causes various strange behaviors as seen in the original discussion.

### Additional Information

See the [original discussion](https://github.com/orgs/micropython/discussions/11901)

### Code of Conduct

Yes, I agree

greezybacon, 2025-05-19:
According to the [TCP RFC](https://datatracker.ietf.org/doc/html/rfc9293#name-half-closed-connections), section 3.6.1, this behavior is recommended by the standard. The server in this case did not read all of the input from the client before it closed the connection. According to the standard, *"If such a host issues a CLOSE call while received data is still pending in TCP, or if new data is received after CLOSE is called, its TCP SHOULD send a RST to show that data was lost."*

The RFC on [TCP Implementation Problems](https://datatracker.ietf.org/doc/html/rfc2525#page-50), section 2.17 on page 50, it explains that *"When an application closes a connection in such a way that it can no longer read any received data, the TCP SHOULD, per [section](https://datatracker.ietf.org/doc/html/rfc1122#section-4.2.2.13)[4.2.2.13 of RFC 1122](https://datatracker.ietf.org/doc/html/rfc1122#section-4.2.2.13), send a RST if there is any unread received data, or if any new data is received."*

The second RFC suggests that CPython on your platform exhibits "Failure to RST on close with data pending"

Josverl, 2025-05-19:
SHOULD is not the same as MUST 

From the RFC 

         *    "SHOULD"

              This word or the adjective "RECOMMENDED" means that there
              may exist valid reasons in particular circumstances to
              ignore this item, but the full implications should be
              understood and the case carefully weighed before choosing
              a different course.

So I think the questions to be weighed is : 
- what would it take for MicroPython to honor this specific SHOULD
- should this partial implementation be documented.

greezybacon, 2025-05-19:
I'm confused. SHOULD is also not the same as SHOULD NOT.

> what would it take for MicroPython to honor this specific SHOULD

You're asking for it to do the opposite of the RFC. It's currently sending the RST as per the *RECOMMENDED* practice of the RFC. If changing the behavior of Micropython/LwIP *"should be understood and the case carefully weighed before choosing a different course"*, in what way is the current implementation incorrect? (other than differing from some other platform.

> should this partial implementation be documented.

What is the partial implementation? It's currently following the RFC AND even the recommended SHOULD bits.

sosi-deadeye, 2025-05-19:
I would simply follow the implementation of the network stack in the Linux kernel. Almost all devices on this planet have network connections to some Linux server. Since this works quite well, the developers seem to have done something right.

_Is there a good reason not to do this?_

Josverl, 2025-05-19:
@greezybacon 

> You're asking for it to do the opposite of the RFC

No , Im just pointing out, same as you, that there are optional parts of that RFC.

> What is the partial implementation

Its according to RFC , but it does not implement all of the Optional parts of the RFC.

And you are correct , the stack already follows the recommendation.

stupid-stove, 2025-05-21:
I can't judge wherever or not this is the "correct" behavior, but as it is diverging from what CPython does, I would suggest to at least put a note in the documentation of the socket API. It took me a while to find out what was happening, so it would be nice to properly document it.
Maybe something along the lines of:

> Note\
> Unlike some other TCP implementations (like Linux), Micropython implements the following optional feature:
> When a call to `close` happens, while data has not been read, a RESET packet will be send to indicate that some data was lost.

(My English isn't that great, so probably somewhat rephrased)

projectgus, 2025-05-22:
I think the MicroPython behaviour is actually subtly different to what's suggested in that RFC. Looking closely at the RFC2525 2.17 ["Trace file exhibiting better behaviour" and "Trace file demonstrating correct behavior"](https://datatracker.ietf.org/doc/html/rfc2525#page-52) the scenarios are where either:

* Our end of the TCP connection receives packets with data after the FIN
* Other end of the TCP connection has sent data that hasn't been ACKed at the time of FIN, due to full window.

In the scenario @stupid-stove captures above, all the received data has been ACKed at the time of the FIN, so we've already told the other side that there's nothing pending. In the lwIP TCP/IP stack MicroPython uses it shows MicroPython calls [tcp_close](https://github.com/lwip-tcpip/lwip/blob/77dcd25a72509eb83f72b033d219b1d40cd8eb95/src/core/tcp.c#L484) and `tcp_close` calls [tcp_close_shutdown](https://github.com/lwip-tcpip/lwip/blob/77dcd25a72509eb83f72b033d219b1d40cd8eb95/src/core/tcp.c#L348) with the `rst_on_unacked_data` flag set, which sends the `RST` because "a RST is sent if the application didn't receive all data (tcp_recved() not called for all data passed to recv callback).".

This makes sense for a fully callback-driven LWIP firmware, but doesn't quite make sense for MicroPython where the TCP receive callback doesn't process any data itself (unless in the Python callback).

Long story short, I think what MicroPython should do here is call `tcp_recvd()` immediately before calling `tcp_close()`, and pass the length of received bytes which have already been ACKed back to the sender but may not have been read into Python (i.e from the viewpoint of the TCP/IP stack they're received, even if not from the point of the "application"). This would resolve this issue, but I think still supports correct RST behaviour according to RFC2525 2.17. I reckon this is also *probably* similar to what a larger operating system like Linux does, with its additional capability for buffering between network interface and application (although I haven't looked into that).

dpgeorge, 2025-05-22:
Side note: MicroPython's lwIP bindings also don't handle the case where a (non-MicroPython) client sends RST before the MicroPython server has a chance to read all the data.  In that case lwIP is abandoning the data upon TCP RST.  See test `tests/multi_net/tcp_client_rst.py`.  This test passes on unix MicroPython but fails on bare-metal-lwIP MicroPython.