micropython/micropython #17075: Syntax Error when using multiline f-strings on RP2040 with MicroPython v1.23.0

kpg141260, 2025-04-04:
### Port, board and/or hardware

RP2040 Raspberry Pi Pico

### MicroPython version

MicroPython Version: v1.23.0 (2024-06-02)
RP2040 Platform: Raspberry Pi Pico
Python Version: MicroPython (RP2040)
Firmware: v1.23.0
Hardware: Raspberry Pi Pico (RP2040)

### Reproduction

```python
msg = (
    f'One ' 
    f'Two ' 
    f'Three'
)

print(msg)
```

### Expected behaviour

The above code should print the concatenated string One Two Three without any syntax errors, similar to how it works in standard Python environments.

### Observed behaviour

```SyntaxError: invalid syntax```

### Additional Information

This issue persists even when re-typing the code, leading me to suspect it might be an issue with the way multiline f-strings are handled on this particular platform.

The issue can be resolved by concatenating the f-strings manually using +:
```python
msg = (
    f'One ' +
    f'Two ' +
    f'Three'
)

print(msg)
```
### Workaround: 
Using string concatenation (as shown above) works, but this is not as elegant and adds unnecessary complexity compared to expected behaviour.

### What else I have tried:
Testing with other Python environments (such as standard Python 3.x) shows that the multiline f-string should work as expected.

This issue does not occur with simpler, single-line f-strings.
Ensuring no invisible characters or incorrect syntax in the code.

### Code of Conduct

Yes, I agree

robert-hh, 2025-04-04:
When pasted directly into REPL it works fine. When pasted with paste mode Ctr-E I get a syntax error. When deleting the white space at the end of the lines it works as well using Ctr-E paste. Using a continuation backslash it works as well.
```
msg = (
    f'One ' \
    f'Two ' \
    f'Three'
)
```

jomasnash, 2025-04-22:
This is not a bug. Micropython implements Python 3.4 and some features of Python 3.5.
You will get the same error in Python 3.4.

kpg141260, 2025-04-23:
Hi, thanks for your reply.In my opinion this should be considered as a bug. But alas, I have a work-around.Kind regards,Peter Gossler

peterhinch, 2025-04-23:
The original script works on the Unix build under V1.25, both when pasted (ctrl-e, ctrl-shift-v, strl-d) and when saved to a file and imported. https://github.com/micropython/micropython/pull/15217 was implemented post V1.23.

jonnor, 2025-10-03:
Seems like this might be fixed?