micropython/micropython #17079: ESP32: Guru Meditation some time after ppp.active(False)

rambo, 2025-04-05:
### Port, board and/or hardware

esp32 port, WROVER

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Generic ESP32 module with SPIRAM with ESP32

### Reproduction

Python code at https://gist.github.com/rambo/af8fed021feba46ce0bba86ea2b9dfaa

A more minimal version is to probably just call net.active(False) after timing out from

```python
net = network.PPP(self.uart)
net.active(True)
await asyncio.sleep(0.5)
net.connect()
while not self.net.isconnected():
    await asyncio.sleep(0.1)
```

Do note that PPP.disconnect -method does *not* exist in ESP32 port for some reason so we cannot call it before calling active(False). 

### Expected behaviour

Micropython should continue running normally forever.

### Observed behaviour

Micropython crashes with Guru meditation.

[mpy_stackraces.txt](https://github.com/user-attachments/files/19615213/mpy_stackraces.txt)

### Additional Information

[mpy_stackraces_resolved.txt](https://github.com/user-attachments/files/19615215/mpy_stackraces_resolved.txt)

See the file for the resolved stack traces.

### Code of Conduct

Yes, I agree

rambo, 2025-04-05:
https://github.com/micropython/micropython/pull/17008 (with relevant changes to the code (remove `.active()` calls, make sure `.disconnect()` is called) seems to solve this issue.

rambo, 2025-04-05:
https://github.com/micropython/micropython/issues/16887 might have same underlying issue

robert-hh, 2025-04-05:
Probably not, since it is a different port and implementation.

dpgeorge, 2025-05-15:
Please try again now that #17138 is merged.  That hopefully solves the issue.