micropython/micropython #17080: build ports/rp2 fail on debian12 with: ERROR: Cannot plug gap greater than alignment - gap 254000, alignment 32

momefilo, 2025-04-05:
### Port, board and/or hardware

rp2

### MicroPython version

Hallo@all i need Help please, i have an Error by Building ports/rp2 on RPi5 Bullseye and x86-64 debian12 without any Modification on Code.

```
momefilo@pi5:~/python/micropython $ git describe --dirty
v1.24.0-443-g3805e65ed
```

### Reproduction

Here is what i do on Rpi5 with raspiOs and a x86-64 with debian12:

```
momefilo@pi5:~/python $ uname -a
Linux pi5 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```
`momefilo@pi5:~/python $ git clone https://github.com/micropython/micropython.git --branch master`
```
momefilo@pi5:~/python/micropython $ git describe --dirty
v1.24.0-443-g3805e65ed
```
`momefilo@pi5:~/python/micropython $ make -j4 -C mpy-cross >> error0.txt`
found no Error no Warning in error0.txt

`momefilo@pi5:~/python/micropython/ports/rp2 $ make -j4 submodules >> error1.txt`
found no Error no Warning in error1.txt

`momefilo@pi5:~/python/micropython/ports/rp2 $ make -j4 >> error2.txt`
shows the following three Erros:

```
[  3%] Performing build step for 'picotoolBuild'
make[4]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird betreten
make[5]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird betreten
make[6]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird betreten
make[6]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird verlassen
make[6]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird betreten
make[6]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird verlassen
[ 14%] Built target errors
make[6]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird betreten
```

```
[  5%] Performing install step for 'picotoolBuild'
make[4]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird betreten
make[5]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird betreten
make[6]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird betreten
make[6]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird verlassen
[ 14%] Built target errors
make[6]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO/_deps/picotool-build“ wird betreten
```

```
[ 98%] Linking CXX executable firmware.elf
ERROR: Cannot plug gap greater than alignment - gap 254000, alignment 32
make[3]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO“ wird verlassen
make[2]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO“ wird verlassen
make[1]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO“ wird verlassen
-e See [1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting[0m
```

### Observed behaviour

```
[ 98%] Linking CXX executable firmware.elf
ERROR: Cannot plug gap greater than alignment - gap 254000, alignment 32
make[3]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO“ wird verlassen
make[2]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO“ wird verlassen
make[1]: Verzeichnis „/home/momefilo/python/micropython/ports/rp2/build-RPI_PICO“ wird verlassen
-e See [1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting[0m
```

### Additional Information

There is no Firmware compiled
On thuesday 01.04.25 it works fine, and on 03.04.25 i have this Problem.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-04-05:
Please see #17073.

The temporary fix is to build and install `picotool` separately.  Eg run:
```
git clone https://github.com/raspberrypi/pico-sdk.git
(cd pico-sdk && git submodule update --init lib/mbedtls)
git clone https://github.com/raspberrypi/picotool.git
(cd picotool && mkdir build && cd build && cmake -DPICO_SDK_PATH=../../pico-sdk .. && make && sudo make install)
```

closed 2025-04-05 by dpgeorge

nkrba, 2025-04-05:
Thank you so much for your detailed responses reg errors like
```ERROR: Cannot plug gap greater than alignment - gap 254292, alignment 32```

Just faced the same and tried your commands above to install `picotool` separately. But after doing this, will the subsequent rp2 build (`make BOARD=RPI_PICO FROZEN_MANIFEST=/path/to/manifest.py`) automatically pickup the separately installed picotool? Or should we give it the location and if so how?

Also during picotool build it complains
```
CMake Error at pico-sdk/lib/mbedtls/CMakeLists.txt:23 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.
```
to resolve which I had to edit `pico-sdk/lib/mbedtls/CMakeLists.txt:23` and change `2.8.12` string to `3.5`. Is this the correct way to resolve?

nkrba, 2025-04-05:
ok best resolution is as mentioned by dpgeorge in https://github.com/micropython/micropython/pull/17073 
edit `GIT_TAG develop` to `GIT_TAG master` within `pico-sdk/tools/Findpicotool.cmake`

momefilo, 2025-04-05:
Thanks very much, the solution from nkrba with:
`edit GIT_TAG develop to GIT_TAG master within pico-sdk/tools/Findpicotool.cmake`
Compiles the Firmware.

dpgeorge, 2025-04-05:
> Just faced the same and tried your commands above to install `picotool` separately. But after doing this, same error remains with the subsequent rp2 build (`make BOARD=RPI_PICO FROZEN_MANIFEST=/path/to/manifest.py`). Will that build automatically pickup the separately-installed picotool? Or should we give it the location and if so how?

It should automatically find it, due to cmake fragments placed in `/usr/local/lib/cmake/picotool`.  Maybe try logging out and in again to your terminal?

Otherwise, follow the instructions in the picotool README.

> ok best resolution is as mentioned by dpgeorge in [#17073](https://github.com/micropython/micropython/pull/17073) edit `GIT_TAG develop` to `GIT_TAG master` within `pico-sdk/tools/Findpicotool.cmake`

Yes that will also work.